dprojects/Woodworking #2: Ready to be added to Addon Manager?

luzpaz, 2022-05-06:
If so, you'd need a [package.xml](https://wiki.freecadweb.org/Package_Metadata) file.

dprojects, 2022-05-07:
[package.xml](https://github.com/dprojects/Woodworking/blob/master/package.xml)

closed 2022-05-07 by dprojects

luzpaz, 2022-05-07:
you can add macros as dependencies if you want...i think

luzpaz, 2022-05-08:
Can you help me understand the odds and ends of how to install this workbench ? It looks like there are a series of macros that it depends on.

luzpaz, 2022-08-19:
Please re-open this ticket as we didn't add this to the Addon Manager. I reiterate  

>Can you help me understand the odds and ends of how to install this workbench ? It looks like there are a series of macros that it depends on.

What are the perquisites ?

Andor, 2023-06-28:
https://github.com/FreeCAD/FreeCAD-addons/pull/260