dprojects/Woodworking #7: Measure Linear Tool Results not able to be deleted

thothloki, 2022-09-21:
I was grabbing some quick measurements for spacing for shelves and was using the Measure Linear Tool. Now, I cannot remove the results. Any way i can?

thothloki, 2022-09-21:
Nevermind. It helps if I put on my glasses and see the "Clear all dimensions" button. Sorry to bother

dprojects, 2022-09-21:
[Part Measure Linear](https://wiki.freecadweb.org/Part_Measure_Linear) is FreeCAD feature, it is not my invention. It opens combo box GUI so it is not easy to find it. You don't see it is already open and active. Also the GUI size not match the window size. This is why I don't like the "combo box" idea and all my tools opens its own GUI. 

I don't use this tool often. Maybe in the future I will create something better, but I am not sure how to draw measurements, with line Draft maybe. I will reopen this issue, until I will find out something better. Let me know how you think this should work meanwhile. 

btw: I like your profile picture, funny sense of humour ;-)

dprojects, 2022-09-24:
There is new tool named [magicMeasure](https://github.com/dprojects/Woodworking/tree/master/Docs#magicmeasure). 
Video: https://www.youtube.com/watch?v=d2FRECuHy2o

To measure space between two shelves you can select 2 vertices in corners. Maybe in the future this will be updated with angle support but for now I don't have good function to draw measurements, I use the FreeCAD one. If in the future I will make new draw function, I will add such thing. 

Let me know if you see anything that could be improved. Now I want to add new report type for BOM, cut-list tool, to get the list of the measurements. I think this will be nice report to have all measurement that can be printed.

closed 2022-09-24 by dprojects