dprojects/Woodworking #8: Undo is not working

magic-k, 2022-09-28:
Woodworking actions are not registered for undo which makes undoing mistakes very hard.

dprojects, 2022-09-28:
Yes, this is true. But the idea behind woodworking tools is to keep it so simple, to not need undo things. You can remove object and quickly create new one. 

I am not sure if some operation can be "undo" especially those Cube -> Pad replace. Probably FreeCAd will not be able to re-create the Cube from Thickness profile. It is more quicker to remove the profile and create new one. 

However, I will keep this issue open for some time, to look closer at this. You have any specific situation that the undo would be needed?

MauricioLucas, 2022-09-29:
This becomes a problem when you have decades of muscle memory.
I have the habit of measuring and working without pencil notes, I do several things at the same time while taking care of my young son. So I go to the kitchen to observe the food that is on the fire, I answer the phone and when I come back my son is already using the keyboard or mouse. No matter what happens, I unconsciously use ctrl+z and lose everything I've done hahahaha.
But don't worry about me, the undo history is a chronic problem with FreeCad.

dprojects, 2022-09-29:
@MauricioLucas Someday you will come and see there is new FreeCAD version written ;-)

The woodworking workbench approach is little different than FreeCAD "parametric" and "undo" approach. The woodworking workbench approach is more like "building from boxes", should be quick and simple. For sure you can "undo" some PartDesign operations.

MauricioLucas, 2022-09-29:
Don't worry, even with some mishap, I'm really enjoying using your Workbench

magic-k, 2022-09-30:
Seems i have to rework my work flow a bit. I am quite similar to @MauricioLucas  so Ctrl-Z is a reflex which unfortunately does a lot of damage in woodworking workbench. But ok, i need to approach the workbench like real woodworking where there's also no undo :-)
Besides that i really love your workbench.

dprojects, 2022-09-30:
> i need to approach the workbench like real woodworking where there's also no undo :-) 

yes, think twice, cut once ;-) 

> Besides that i really love your workbench.

Thanks. It is always very nice to hear that someone like the work you do. The "undo" is not priority for the workbench but I will look into that if I finish the support for [irregular shapes](https://forum.freecadweb.org/viewtopic.php?f=22&t=72179). So, I leave this issue open until this time. 

I usually use CTRL-Z at Kwrite so I know it might be useful. At FreeCAD I click CTRL-S very often and make backup of the file ;-)

MariwanJ, 2022-10-01:
Adding undo is a mater of adding two lines. Why wouldn't you add that? 
You can look at my simple tools that uses the undo mechanism.. It is easy to implement .. just a copy-paste

dprojects, 2022-10-01:
> Why wouldn't you add that? 

Hello Mariwan here LOL ;-) because I don't use undo? for me it is not needed feature. If I make mistake I usually remove object and create new one. Backup is always essential. This is simple with PartDesign objects, here is sample how you can quickly remove object and create new one: [3D furniture designing - construction profiles](https://www.youtube.com/watch?v=5hXMFAxXQag) of course not all features works like this...  

> Adding undo is a mater of adding two lines.

I am pretty sure this will be more complicated and FreeCAD will not be able to re-create objects... 

> You can look at my simple tools that uses the undo mechanism.. It is easy to implement .. just a copy-paste

but I will look at your code and at this issue closer, I will play with CTRL-Z with all features and will see what will happen ;-) kernel panic? ;-) first I need to find [some kind of workaround for 0.20](https://forum.freecadweb.org/viewtopic.php?f=22&t=72250), and next I will look at it...

dprojects, 2022-11-24:
For all PartDesign features you can remove last operation. If you press CTRL-Z sometimes the Cube is restored. So I don't see issue here. Please be more specific.

closed 2022-11-24 by dprojects