dprojects/Woodworking #9: Your License

MariwanJ, 2022-10-01:
HI, 
I am not expert regarding Licenses, but why don't you use the same License as freecad? 
They told me that if you ever wish to join your WB to FreeCAD you have to have the same License.
What do you think?

dprojects, 2022-10-01:
What I think? about typos? I don't plan make more typos so typo-snake might be disappointed ;-) From law perspective it is not true you need to have LGPL to be part of AppImage.

FreeCAD is patchwork with about 1000 different licenses. Unpack AppImage and search for license file so you will see how many different licenses there are, many of them are MIT. Personally, I don't like the LGPL attitude. MIT allow for much more than LGPL but still keep ownership and being author of the code.

MariwanJ, 2022-10-01:
ok

closed 2022-10-01 by MariwanJ