dprojects/Woodworking #10: Wrong button texts in dialogs in german locale

magic-k, 2022-10-04:
I run FreeCAD on a german chromebook. When opening a dialog (e.g. MagicManager) instead of button texts like ">" or "<" i see ":" and ";". See screenshot of MagicManager:
[image]

dprojects, 2022-10-04:
There is also other issue I see. The sizes of the buttons are to small so they not show the text correctly. I don't have chromebook, so I am not able to reproduce the problems. However, this might be problem with theme. The button looks better, but they probably have have too big pad around the text. Try other theme... if there is also problem with ">" or "<".

magic-k, 2022-10-05:
Chromebook means basically a debian inside a container. Could be a problem with the x11/wayland window manager. I'm also using the Appimage version. But i don't see this in other dialogs.
At least the problem with ">" and "<" went away after changing the theme.
[image]

dprojects, 2022-10-05:
But there is still problem with extra padding added around the text without resizing buttons. I don't want to make the GUI to be half of the screen. There always will be this type of problems. Imagine how many languages are around the world... some languages have much longer translations.  So, this is very hard to create GUI to be looking good at any theme and any language and also will not be huge... however maybe little GUI resize would be accurate here, because I see there are not so many letters cut. I will resize the GUI little bit but I don't have to test it. 

Do you observe anything like that at other GUIs?

dprojects, 2022-10-07:
I redesigned a little bit the GUI of magicManager. Now there is single button to create panel. I always make mistake and clicked not the correct one ;-) Also the GUI is little more wide and the buttons as well. Check if this is fine now.

magic-k, 2022-10-18:
It's way better. There are still a few texts that are cut, but nothing crucial.

dprojects, 2022-11-24:
There is not problem with FreeCAD themes. Probably the issue is related to the "incorrectly created" theme. The theme resize fonts but not resize widgets like buttons. Please install valid theme.

closed 2022-11-24 by dprojects