dprojects/Woodworking #11: I can't move a selected object with the tools

skredders, 2022-10-14:
I use WIN10, FreeCad 0.19 and Woodmaker 0.20

When I select a part I can't move this with the tools. I receive always the message "Please select valid object to move" even if the part is highlighted. What's wrong?
[image]

dprojects, 2022-10-14:
What do you mean by Woodmaker 0.20 ? It is recommended to use at least FreeCAD 0.20 stable: 
https://github.com/dprojects/Woodworking/releases/tag/0.20

Moving via Part is working for me and I don't have Windows so I am not able to reproduce this problem. However if you want to debug this on your own You can comment try block at panelMove function inside MagicPanels library... and see where it stucks... 

btw: For single objects it is better to move the object via its own placement not via container placement. Especially if you want to apply dowels or drill holes later, or even measure. If you move object via container the dowels should also be in that container...

closed 2022-10-14 by dprojects

skredders, 2022-10-14:
I have installed freecad 0.20.1 and it works so far.
Thank you for your advice.

dprojects, 2022-10-14:
If you use FreeCAD 0.20.1 version you can try the latest master branch. There is new feature to create panel from vertices [3D furniture designing - corner kitchen cabinet](https://www.youtube.com/watch?v=fOh7bQfgkcg). This feature use Draft Sketch conversion that should work at FreeCAD 0.20.1. 

If you want better stability make sure your FreeCAD version match the Woodworking workbench version.