dprojects/Woodworking #12: Windows Testing ... I might be volunteering (on issues found so far ... breath easier)

ShamanTcler, 2022-11-01:
I see that the releases are primarily Apple/*nix based. But I am running Win 11 (insiders edition)

I am also running Free 0.21 (Beta code at the moment) (Python 3.8)

I installed the master branch of the workbench in: C:\Users\*****\AppData\Roaming\FreeCAD\Mod

Then proceed to load an example file and it appeared to load just fine.

I have been using the FreeCad Path workbench a lot, it will be interesting to see how Path deals with some of your models

Oh and I read $150 PLN .... would that be Złoty? I have family in Mława... though I am in the USA

dprojects, 2022-11-01:
I have only laptop with Xubuntu so I am not able to say if this works at other operating systems. The "certify" means I have tested all features... I think this can be tested by community, but as long I don't have other system I can say it is "certified". It can be certified by community, why not. I don't know, maybe this should be special topic at FreeCAD forum or maybe here as issue? 

As far as I know it is better to use AppImage than native installation at Windows or macOS.

dprojects, 2022-11-24:
There is information about other platforms in "Certified Platforms" readme section: 
https://github.com/dprojects/Woodworking#certified-platforms

closed 2022-11-24 by dprojects