dprojects/Woodworking #14: If you cut a panel, it won't show up in the cut list without the base panel being shown

akdombrowski, 2023-01-27:
Love the workbench! Makes my life much easier.

If you use the cut list with the visible objects option turned on so only visible objects show in the cut list, cut objects don't show up in the cut list if the base panel isn't visible.

For example, I create a panel and a dowel. If I cut the panel with the dowel, the panel and the dowel are no longer visible. I then use cut list and choose to only show visible objects, that cut object doesn't show in the cut list.

I've got a project with panels with tons of cuts, I'd have to go and make all the base panels visible in order for them to show up in the cut list. If you could show the panel dimensions if the parent cut object is visible, that'd be helpful.

Thanks!

akdombrowski, 2023-01-27:
[image]

dprojects, 2023-01-27:
> If you use the cut list with the visible objects option turned on

If you use Cut in the FreeCAD way you should rather use **Parent + Base**. 
More about visibility options: [Docs - visibility](https://github.com/dprojects/getDimensions/tree/master/Docs#visibility)

I guess this is the same issue as this one here: [getDimensions issues 12](https://github.com/dprojects/getDimensions/issues/12)
I believe it has been solved, and the solution is Parent + Base. However if this not work for you, provide some simple file example, single Cut, not huge project. So, I will try to figure out if there is still some scenario not covered.

dprojects, 2023-01-27:
However, I see if you have many Cuts the naming conventions starts to be annoying, like "Cut, Cut, Cut... " zillion times is not good ;-) I am not sure how to approach that and fix it and keep the structure informative at the same time. This has been solved with [getNestingLabel](https://github.com/dprojects/Woodworking/blob/1d277b1ff4aba07b8468330dc89ab39bcf9f340e/Tools/MagicPanels/MagicPanels.py#L2506) function for multiple copy, but I am not sure if this would be good approach for Cut.

akdombrowski, 2023-01-27:
Ah, I didn't see that option in the cut report options. Is that in the development branch?

dprojects, 2023-01-27:
Yes, this is in master branch. If you are using magicCut you should rather use visibility Off or Inherit mode. For Cut listing you can list all because all the hidden elements are cloned and the copies are turned off at report with special attribute BOM. I need to fix the "Cut,.... " labeling ;-)

dprojects, 2023-01-27:
I [fixed the long labeling for Cuts as well](https://github.com/dprojects/Woodworking/commit/f0149028bc43a30f3113c446c497a6ace64ae968), so this should no longer be disturbing. The previous one was little more informative, especially if you already have ", " at label but there is numbering so the nesting level should also be recognized.

akdombrowski, 2023-01-28:
Thanks! I'll try out the master branch. What version of FreeCad should I use?

closed 2023-01-28 by akdombrowski

dprojects, 2023-01-29:
> Thanks! I'll try out the master branch. What version of FreeCad should I use?

If you want only getDimensions macro update you can upload only the macro. 
If you want nesting labeling improve you have to download master branch of workbench. 
At 0.20 FreeCAD there is bug with Sketch create from vertices. Maybe 0.20.2 have it fixed I don't know. The weekly latest AppImage should work but I didn't test it. You can download the [recommended weekly](https://github.com/dprojects/Woodworking#certified-platforms). 

Regarding your problem, what is your project process? Is there any field to improve? I see you are using Cylinders for Cut. Normally, drilling feature is designed with PartDesign Hole objects not with Boolean Cut. But I saw that other people use this way too. Maybe this is worth to improve?

akdombrowski, 2023-01-29:
Thanks, things seem to be working so far on 20.2 with the master branch.

I was creating dowels using the magic dowels feature. Then, I was cutting my panels with those dowels. I want to have dowel dimensions to print out. And sometimes I have dowels going through multiple panels.

dprojects, 2023-01-29:
> I was creating dowels using the magic dowels feature. Then, I was cutting my panels with those dowels. I want to have dowel dimensions to print out. And sometimes I have dowels going through multiple panels.

If you cut many panels with single dowel, you can use: [magicKnife](https://github.com/dprojects/Woodworking/tree/master/Docs#magicknife). This tool creates copies for each Boolean Cut Tool element so you don't need to change default settings at getDimensions tool later. The copies will be turned off at the report and the original ones will stay in the same place and will be visible at report.

akdombrowski, 2023-01-29:
It's multiple dowels positioned throughout. I stopped doing the cutting, but I uploaded it here if you want to take a look and have any suggestions:
[https://github.com/akdombrowski/outdoorTVStand9/blob/main/outdoorTvStand9.FCStd](url)

dprojects, 2023-01-30:
> I stopped doing the cutting

I downloaded your all TVStand9 repository. My comments: 

- At png files I saw green measurement from vertex to edge, I don't remember if this is available at 0.20 stable version or it was added later but for sure at the master branch you can measure Vertex -> Edge and also Vertex -> Face
- You are using hide method, this is little disturbing for features, because some features parse all FreeCAD objects, so the "hidden" might be parsed too. There is such visibility option at cut-list report, but rather avoid it. Better make backup of your project and remove not needed objects. So the file will be smaller. 
- For cutting dowels, it is hard to find the right dowels that belongs to the panel. So, I added new feature for that. It is called [cutDowels](https://github.com/dprojects/Woodworking/tree/master/Docs#cutdowels) and it allows you to select only panel without searching the dowels. The tool will cut out the exact dowels automatically. This feature is visibility sensitive so it is working with you file sample with hidden structure. 

Video: https://www.youtube.com/watch?v=Oogs8LqkReQ

dprojects, 2023-02-02:
Just to let you know:

- I saw at your model bigger vertices. So, I added new tool called [showVertex](https://github.com/dprojects/Woodworking/tree/master/Docs#showvertex). This tool work with the same way as making panels transparent, so you can toggle all Vertices bigger and change color to red quickly for a while and change the size and color to normal. So you can keep the model good looking without disturbing bigger vertices.
- Also I added new tool to help vertex selection. It is called [selectVertex](https://github.com/dprojects/Woodworking/tree/master/Docs#selectvertex). This tool allow you to select nearest vertex if you select edge or face.

Video for the features: [Helping Vertex selection](https://www.youtube.com/watch?v=qSsua04AKg8)

Now you designing way should be much more simpler and faster. Thanks and Good Luck !

akdombrowski, 2023-02-02:
This is amazing! Thank you, I will try it out.