dprojects/Woodworking #15: Lost reference values when using functions like Copy panel XYZ

akrly, 2023-02-18:
Hi,

First of all amazing workbench. I've attempted delving into FreeCAD many times, but got put off by the learning curve and went back to Fusion360 (Had an education licence for years). However, I decided to give it another go and this workbench helped get through the initial pain points.

I am working now on developing a model for a piece of furniture and I'd like it's dimensions to be driven by a table. I realised when I use some the provided functions such as copy panel xyz, that only the dimensions' value is translated to the new object. Same thing happens with routerStraight2 and so on.

Now I gather that the goal of this workbench is to quickly model in woodworking plans I still wonder could there be a way to propagate driven dimension throughout these functions?

I've uploaded an example [here](https://github.com/akrly/parametric_drawer). The photos below meant to illustrate the issue. 

[image]
[image]

akrly, 2023-02-18:
Okay, so I have to partially retract my post as just tried to simple Crtl+C, Ctrl+V the body and that kept the driven dimensions. I guess this shows how new am I to the program. :)  Apologies

The issue still stands with routerStraight2 though. Looking at the model tree it does invoke a panel2pad. So that too.

dprojects, 2023-02-19:
> First of all amazing workbench. I've attempted delving into FreeCAD many times, but got put off by the learning curve and went back to Fusion360 (Had an education licence for years). However, I decided to give it another go and this workbench helped get through the initial pain points.

Thanks, very nice to hear that. I don't use other CAD software, and never used, but what I saw at YouTube the Fusion360 is much more advanced than FreeCAD. There is elegant 3D Sketching and almost automatic parametrizations. I would never look at FreeCAD as a replacement of Fusion360 or SketchUp, those programs are much more advanced. 

> I am working now on developing a model for a piece of furniture and I'd like it's dimensions to be driven by a table. I realised when I use some the provided functions such as copy panel xyz, that only the dimensions' value is translated to the new object. 

By default the expressions are not copied. But I think this is good issue to consider. The expressions are stored in single place ExpressionEngine, so I need to check if they also can be copied. 

However, I would rather recommend little different strategy. You can use Clones and if you really want Spreadsheet with variables to change, you can make parametric only the first object, all other Clones will be automatically  changed. 

I use this approach at the last video: [Align to curve](https://www.youtube.com/watch?v=fbJV_SEuNLg)

> Same thing happens with routerStraight2 and so on.

If you use Clones, any operation will be automatically populated to the Clones. This is some kind of Direct Parametric Modeling, you don't need to worry about expressions if you use Clones or Links, but Links are good only for view, you can't drill holes thru Links.

akrly, 2023-02-19:
Indeed Fusion and other commercial CAD software has a lot of bells and whistles, however they come with the price of having to accept their UI changes and forcing you into their cloud infrastructure both of which can seriously affect the workflow and past projects. So I think FreeCAD can be a good alternative when stability is a key point especially for non-commercial use.

Thanks for the tip. I'll try out that strategy and report back in a bit.

dprojects, 2023-02-19:
> Indeed Fusion and other commercial CAD software has a lot of bells and whistles, however they come with the price of having to accept their UI changes and forcing you into their cloud infrastructure [...]

FreeCAD make changes too, in my opinion too much. I see paid CAD software more for professional corporate job. FreeCAD is good, for example, if you want to make furniture for your garage or home that costs $200. In this case buying so expensive software is little pointless. Also if you are developer FreeCAD is good option because you can do some features for your purpose, if you don't need so advance modeling as for example SketchUp has. Also if you have slow hardware without Windows as I do, the FreeCAD might be better option. So, in my opinion FreeCAD should still release 32-bit Windows version. 

Many people comes to FreeCAD and see it as a replacement of paid CAD software in general meaning, for professional job. I think this approach may only end up with disappointment.

> Thanks for the tip. I'll try out that strategy and report back in a bit.

Yes, play with Clones a little bit. I think this might be the way you are looking for without writing the math formulas each time.

dprojects, 2023-02-21:
This issue has been explained at: [Parametrization - the basics](https://www.youtube.com/watch?v=NjFl8W6-Vjg)

closed 2023-02-21 by dprojects