dprojects/Woodworking #17: user might need to clone repo to $user_dir/Mod

ergodicbreak, 2023-03-24:
Hello, thanks so much for this workbench! I installed FreeCAD because I wanted to try modeling a project for the first time. I was so happy to see this workbench available. 

I wanted to make a note that the install instructions don't mention needing to clone the repo to the `Mod` directory (in my case on Manjaro, located in `/home/$user/.local/share/`. At least in my case I had to do this before the workbench could be loaded. 

I'm very excited to start learning FreeCAD so thanks a lot for your videos too! 

Cheers, George

dprojects, 2023-03-24:
Thank you for the kind words. The installation process is described at: https://github.com/dprojects/Woodworking#installation

There is sentence: "If there is no Mod folder, create it." Let me know if this is it.

ergodicbreak, 2023-03-24:
Thanks, I just misread it. I thought the direction "to get location of Mod folder, run ...." meant that returned the Mod folder -- not its parent directory.

closed 2023-03-24 by ergodicbreak