dprojects/Woodworking #19: Tools missing?

ansgar02, 2023-03-28:
Looking for a design tool for my hobbyist woodworks I found and start to love the WW WB (and FreeCAD)
After completing my first project by more or less try and error and using the popup explanations/help text I started to dig into the documentation more systematically (and some more tutorial driven experience with basic FreeCAD).
In the doc I found some interesting features/tools like panel2angle or cornerblock (in the construction section), but I'm unable to identify the funcionality in th UI (neither default layout nor Toolbar configuration etc.).
Do I have a blind spot or is there a way for me to enable those tools?
Please find my environment below!

Many thanks for this great work!
Cheers
Ansgar

I'm using:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly4 0.12.6
  * Curves 0.6.8
  * dodo 1.0.0
  * dxf-library
  * fasteners 0.4.55
  * freecad.frametools 0.1.0
  * Help 1.0.3
  * kicadStepUpMod 10.17.5
  * LCInterlocking
[/code]

dprojects, 2023-03-29:
> panel2angle or cornerblock (in the construction section), but I'm unable to identify the funcionality

The panel2angle is described here: [panel2angle](https://github.com/dprojects/Woodworking/tree/master/Docs#panel2angle)
The cornerblock is described here: [cornerBlock](https://github.com/dprojects/Woodworking/tree/master/Docs#cornerblock)

Watch the videos at YouTube linked there. The panel2angle allows to create custom profile just by selecting faces. You select faces that should be removed. The cornerBlock allows to create support in the table leg in the corner. To create it you have to select single vertical edge.

I am not familiar with Windows so I can't help here. If the tools not run, there might be problem with FreeCAD version or other workbenches impact.

ansgar02, 2023-03-29:
Thank you for your timely response! 
Actually it looks like I had an incomplete/corrupted version of the WW WB (date in package.xml: 20220905).
Creating a clean clone from the master and installing this immediately shows the respective icons. May be, this resolves a few other things, which I had thought were user errors (although there a still plenty of those :-))

Thanks again!
Ansgar

dprojects, 2023-03-29:
> user errors (although there a still plenty of those :-))

Thank you for kind words. Feel free to open new issue with errors output if you find something incorrect.

closed 2023-03-29 by dprojects