dprojects/Woodworking #20: Some panels not working with Edit -> alignment

ergodicbreak, 2023-04-01:
I have kind of a weird issue I don't understand.

I tried to align a panel with a sketch with Edit -> Alignment, and while the sketch was in the fixed object part of the Alignment window, there was nothing in the moveable object window.

However if I create a new panel, I **can** align that panel with the sketch. 

Do you know of some condition that would happen where a panel would become un-alignable?

dprojects, 2023-04-01:
> Do you know of some condition that would happen where a panel would become un-alignable?

If the panel is rotated. Please provide FreeCAD file sample and more description step by step. I am not sure what scenario you mean.

ergodicbreak, 2023-04-01:
Thanks @dprojects, I'll try to recreate the issue in an easy to see way.

dprojects, 2023-04-05:
should I close this issue?

ergodicbreak, 2023-04-05:
Sure we can close for now, this isn't a critical blocking issue for me and I haven't had time to recreate a test case.

closed 2023-04-05 by ergodicbreak