dprojects/Woodworking #21: Few questions

supernaturaljoe, 2023-04-05:
Hello,
First - great workbench. Very helpful for creating cabinets and exporting BOM.
I have 2 questions though.
I've noticed that when I create a panel Lenght is in X direction and Width is in Y direction which is ok but when I export this detail toCut spreadsheet is show the width first and the length after ( I have attached a screenshot)
[image]

Even if I rotate the panel still shows it this way. In general it is not a big deal because by default I treat the grain direction of the material in Length directions and when I export the BOM list just switch both columns in Excel.
So is there a way to make the workbench to switch both columns and show Length first and Width second or not? Again not a big deal though.
And the second question.
If I place a texture or colour on a face is there a way to recognize this face and shows it in the toCut spreadsheet as different face finish.
For example: I have 3 panels and 1 of them has wood finish applied. And would be nice if in the BOM list to show 2 panels in default + one with wood finish.
Oh, sorry :) one request - can you make the command "edge to straight" to have uneven cut (at the moment 1/4, 1/3, 1/2).. For cabinets usually the cut half of the material width + depth of the back (HDF for example is 2.5mm so I cut the back side 9mm width and 3mm deep to hide the HDF back).
Have a great day.

dprojects, 2023-04-05:
> Hello, First - great workbench. Very helpful for creating cabinets and exporting BOM. I have 2 questions though. I've noticed that when I create a panel Lenght is in X direction and Width is in Y [...]

Thank you for kind words. The dimensions are sorted, so they not match the property window order. This is mostly because the thickness is in separate column. So, if Length would be 18 mm this would be the thickness. 
I would rather recommend new report type. The columns would be named Length, Width, Height - without thickness column and without thickness summary. Let me know if this is what you are looking for.

> If I place a texture or colour on a face is there a way to recognize this face and shows it in the toCut spreadsheet as different face finish. For example: I have 3 panels and 1 of them has wood [...]

The edgeband information can be generated via [e - report type](https://github.com/dprojects/getDimensions/tree/master/Docs#e---report-type). But there is used color not texture. Texture is applied to the whole object not to the face. So it is not possible to determine texture only for face. 

> Oh, sorry :) one request - can you make the command "edge to straight" to have uneven cut (at the moment 1/4, 1/3, 1/2).. For cabinets usually the cut half of the material width + depth of the back [...]

Can you provide more detail on that? any FreeCAD sample file?

supernaturaljoe, 2023-04-06:
__So, if Length would be 18 mm this would be the thickness.
I would rather recommend new report type. The columns would be named Length, Width, Height - without thickness column and without thickness summary. Let me know if this is what you are looking for._

Not exactly but I just swap the columns with dimensions in Excel. I have attached you a FreeCad file and 2 screen shot what I meant.

Also, there is another FC file with a sample cabinet that show what I meant about the cutout.

[samples.zip](https://github.com/dprojects/Woodworking/files/11170691/samples.zip)

dprojects, 2023-04-06:
This is a little problematic approach because Pads don't have Length, Width and Height. 
For Pads, and other PartDesign objects, there is only Length and constraints.

dprojects, 2023-04-06:
> Also, there is another FC file with a sample cabinet that show what I meant about the cutout.

Regarding the sample texture request. The texture is applied at object not face. [Material at FreeCAD](https://wiki.freecad.org/Material) is little different thing than Texture. 

If so I would rather create new report with texture. But what kind of information about texture you would like to see? Link? Name? I am not sure if this is possible to determine grain direction directly from picture, from the filename it would be possible.

supernaturaljoe, 2023-04-07:
I was pretty sure that the texture won't be recognizable but wanted to ask :).
That you for the help.
if you are planning to insert a basic models (such as PVC kitchen legs, handles etc) feel free to contact me and will be more than happy to send you some of my models.
(available at the moment PVC kitchen leg, undermount kitchen sink, aluminium handle profiles)
Have a great day.

dprojects, 2023-04-07:
For grain direction instead of texture, it is better to use [grain direction marker](https://github.com/dprojects/Woodworking/tree/master/Docs#grainh)
For edgeband it is better to [use colors](https://github.com/dprojects/getDimensions/tree/master/Docs#edgeband). 
Texture should rather be used only for model preview.

Let me know if this topic can be closed or there is still open issue.

supernaturaljoe, 2023-04-10:
Oh, cool. New features. Just updated the workbench.
Yes, topic can be closed. 
Thank you for the help :)

closed 2023-04-12 by dprojects