dprojects/Woodworking #22: magicManager error or abuse?

ansgar02, 2023-04-19:
I'm trying to use the MagicManager to create a panel between two panels (beams) (named "QT Oben Hinten" and "QT Oben Vorne" in container "QT Oben" with respective faces "Face5" and "Face6" (see attached FCStd file). When I do so, the Report View shows the following lines:
22:12:14  Traceback (most recent call last):
22:12:14    File "C:\Users\Ansgar\AppData\Roaming\FreeCAD\Mod\Woodworking\.\Tools\MagicPanels\magicManager.py", line 739, in setMode
22:12:14      self.setNextPanel()
22:12:14    File "C:\Users\Ansgar\AppData\Roaming\FreeCAD\Mod\Woodworking\.\Tools\MagicPanels\magicManager.py", line 821, in setNextPanel
22:12:14      self.previewPanel()
22:12:14    File "C:\Users\Ansgar\AppData\Roaming\FreeCAD\Mod\Woodworking\.\Tools\MagicPanels\magicManager.py", line 672, in previewPanel
22:12:14      self.projectPanelBetween(self.gPanelArr[self.gPanelIndex])
22:12:14    File "C:\Users\Ansgar\AppData\Roaming\FreeCAD\Mod\Woodworking\.\Tools\MagicPanels\magicManager.py", line 607, in projectPanelBetween
22:12:14      if L <= 0 or W <= 0 or H <= 0:
22:12:14  TypeError: '<=' not supported between instances of 'str' and 'int'

my environment: 
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32552 (Git)
Build type: Release
Branch: master
Hash: e2b3a5d2f791fb78a61daac9bdab374058f682f2
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly4 0.12.6
  * Curves 0.6.8
  * dodo 1.0.0
  * fasteners 0.4.55
  * Help 1.0.3
  * LCInterlocking
  * Woodworking 0.21.30486
[/code]

my debugInfo shows a half green/red bug with "Tests failed: qApp"
Which FC version should be used with this version of the WB?

Am I using this feature the wrong way?

In the attachement please find my FCstd file and the screen shot with the report view .

Cheers
Oerly.

[image]
[WB004.01FCStd.FCStd.zip](https://github.com/dprojects/Woodworking/files/11277404/WB004.01FCStd.FCStd.zip)

dprojects, 2023-04-20:
> I'm trying to use the MagicManager to create a panel between two panels (beams) (named "QT Oben Hinten" and "QT Oben Vorne" in container "QT Oben" with respective faces "Face5" and "Face6" (see attached FCStd file). 

This is caused because those panels are rotated. If you want to rotate elements put them into container and rotate container instead of rotating the element. The workbench is not ready for single elements rotation, this is mostly caused by problem with determining the plane if the element is rotated. 

> my debugInfo shows a half green/red bug with "Tests failed: qApp" Which FC version should be used with this version of the WB?

The FreeCAD 0.20 stable version has qApp defined but the 0.20 workbench not has so much features.  
Currently there is no FreeCAD version with defined qApp, see more:
https://forum.freecad.org/viewtopic.php?p=676143#p676143
I am waiting for FreeCAD 0.21 and will decide if he qApp test make sense. 

I will keep the issue open because maybe I will look into rotation problem more deeply finally.

ansgar02, 2023-04-20:
Thank you for your swift response - and well - I remember having read about this advice but actually following it sometimes needs another round... Placing those in containers made the magicPanels work as expected.
Many thanks again!

Not sure I understood your FC version recommendation. As I am on MS-W10 (knowing your base is Linux) should I better use the official FC 020 version or the 0.21 version (which I was using when encountering the above issue)?

Ansgar

dprojects, 2023-05-08:
Hello, I made small [rotation improvements](https://github.com/dprojects/Woodworking/commit/fa1dbc8177cb83fc44d174e0dff6187120092995). Now the direct object rotation is better supported. You can create panel at face for rotated object.

However, creation panel between rotated objects is still not covered. This require more advance calculation in relation to the anchor. I am not sure if this will be covered but the rotation improvements will be continued.

ansgar02, 2023-05-09:
Hello,

 

many thanks for addressing this – and for the notification.

I will look into it in the following days.

 

Cheers

Ansgar

dprojects, 2023-05-09:
I will keep this issue open until the rotation feature will be fully finished. At the end I will try to find out the way to support also the panel between rotated objects. Now if you want to create panel between rotated objects without containers you should rather use [panel from vertices option](https://github.com/dprojects/Woodworking/tree/master/Docs#magicmanager).

dprojects, 2024-02-04:
Hello again, I was looking at this problem again, and I am not sure for adding additional option to magicManager. 
Generally panels should be rotated via LinkGroup container, but if you have single rotated panel and you really don't want to use containers you can always use panel from vertices. At your example there is not need to rotate panel at all. 

I am not sure it is worth to keep this issue open? Feel free to re-open this issue if you have questions. Thanks !

closed 2024-02-04 by dprojects

ansgar02, 2024-02-04:
Hi Darek

 

Thank you very much for looking into this – again.

Thank you very much – also for creating this great tool at all!

 

Best regards,

Ansgar