dprojects/Woodworking #25: toCut uses template already used by TechDraw causing serious issue, when the toPrint is deleted

przemhb, 2023-07-29:
In my project I have a few technical drawings created in the TechDraw. I have noticed, that when I run Woodworking's getDimensions function then the resulting toPrint uses the same template as the first TechDraw drawing. As a result, when I delete the toPrint with "its" template, then the template gets removed from the TechDraw page. It causes serious issue. Missing template cannot be added, project has error, cannot be recomputed and properties are no longer displayed. Furthermore the TechDraw cannot be even deleted to start from scratch. Recovering from the issue is really hard.

[image]

przemhb, 2023-07-29:
FreeCad 0.20.2, Woodworking 0.20.

dprojects, 2023-07-29:
Hi, can you attach file sample? No real-life project, just sample to reproduce the problem, I will look into that later what is going on there.

przemhb, 2023-07-29:
Here you have it:

[template-issue.zip](https://github.com/dprojects/Woodworking/files/12207787/template-issue.zip)

Run getDimensions with default settings and then delete template from the "toPrint" and notice it gets deleted from the "01.01'A Blat" technical drawing. Apparently "toPrint" links to already existing template instead of creating a new one.

dprojects, 2023-07-30:
Has been fixed by: [fix template overwrite ](https://github.com/dprojects/getDimensions/commit/9efb90c2ecfac136403a2361c65e06609ac21575). Thanks.

closed 2023-07-30 by dprojects