dprojects/Woodworking #29: Running the Python command 'magicMove' failed

genlinjiao, 2023-10-24:
hello, I am learning the woodworking bench macro and encounted an issue. The error message is following this message. My system is Ubuntu 22.04.3 LTS. Please help to find a fix. Many thanks.
Running the Python command 'magicMove' failed:
Traceback (most recent call last):
  File "/home/genlin/.FreeCAD/Mod/Woodworking/loadTools.py", line 353, in Activated
    __import__(module, globals(), locals(), [], 0)
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/genlin/.FreeCAD/Mod/Woodworking/Tools/MagicPanels/magicMove.py", line 1023, in <module>
    showQtGUI()
  File "/home/genlin/.FreeCAD/Mod/Woodworking/Tools/MagicPanels/magicMove.py", line 16, in showQtGUI
    class QtMainClass(QtGui.QDialog):
  File "/home/genlin/.FreeCAD/Mod/Woodworking/Tools/MagicPanels/magicMove.py", line 65, in QtMainClass
    gCrossCorner = FreeCADGui.ActiveDocument.ActiveView.getCornerCrossSize()

getCornerCrossSize

Tamusk, 2023-10-26:
Woodworking freecad 0.21.1 não carrega

dprojects, 2023-10-27:
> Woodworking freecad 0.21.1 não carrega

To be honest, I am not surprised that FreeCAD 0.21.1 is broken :-) 

The FreeCAD 0.21.0 didn't pass tests so this why I wasn't able to release stable version for Woodworking workbench. The FreeCAD 0.21.1 is also broken, there is no template Width and Height so the [getDimensions](https://github.com/dprojects/getDimensions) report cannot be positioned in the center of the TechDraw page. 

We will see if the FreeCAD 0.21.2 will be better or not, if so I will release stable version for Woodworking workbench based on FreeCAD 0.21.2. If not maybe I will decide to release the stable version based on [FreeCAD weekly builds 30486 2022-09-29 AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-30486-2022-09-29-conda-Linux-x86_64-py310.AppImage).

dprojects, 2023-10-27:
Wrong installation, not supported versions, not code related issue.

closed 2023-10-27 by dprojects