dprojects/Woodworking #30: magicMove error

mad-max-traveller, 2023-11-09:
I'm trying to use the magicMove tool, but I'm getting an error:

```
'Gui.View3DInventor' object has no attribute 'getCornerCrossSize'Running the Python command 'magicMove' failed:
Traceback (most recent call last):
  File "/home/maximus/.FreeCAD/Mod/woodworking/loadTools.py", line 353, in Activated
    __import__(module, globals(), locals(), [], 0)
  File "/tmp/.mount_FreeCAawP184/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/home/maximus/.FreeCAD/Mod/woodworking/Tools/MagicPanels/magicMove.py", line 1023, in <module>
    showQtGUI()
  File "/home/maximus/.FreeCAD/Mod/woodworking/Tools/MagicPanels/magicMove.py", line 16, in showQtGUI
    class QtMainClass(QtGui.QDialog):
  File "/home/maximus/.FreeCAD/Mod/woodworking/Tools/MagicPanels/magicMove.py", line 65, in QtMainClass
    gCrossCorner = FreeCADGui.ActiveDocument.ActiveView.getCornerCrossSize()

'Gui.View3DInventor' object has no attribute 'getCornerCrossSize'
```

Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.10.8
Qt version: 5.15.4
Coin version: 4.0.1
OCC version: 7.6.3

yaroslav-shumlianskyi, 2023-11-12:
@mad-max-traveller, I had the same error. After updating FreeCad to 0.21 it is gone.

mad-max-traveller, 2023-11-14:
Upgraded to 21.1 and the problem went away.