dprojects/Woodworking #31: The data in the showSpaceModel window is poorly visible if dark mode is used.

mad-max-traveller, 2023-11-09:
The data in the showSpaceModel window is poorly visible if dark mode is used. This is also the case in other tools similar to showSpaceModel.

[image]

[image]

dprojects, 2024-01-16:
any idea how to fix it? you can always invent theme color which will be not visible, I can only adjust for base themes

dprojects, 2024-01-19:
I changed the background into red, so now should be fine with classic and dark theme.
Feel free to re-open this issue if you think this should be changed. Thanks!

closed 2024-01-19 by dprojects