dprojects/Woodworking #32: Magic Drill overwrites parametric panel dimensions

flashydave, 2023-11-29:
If you use a spreadsheet to size a panel using parameters and then add holes through the magic drill then the sizes that were defined as formulas get replaced by the current size values when the part is created. 

ie you lose all the elements of the design and need to edit these in manually as constraints on the sketches or attachment coordinates.

The magic drill script should preserve any parametric entries when the part and sub-components of it are created

dprojects, 2024-01-16:
I guess this issue is not only related to single tool, I think this should be general testing with expressions. 
Well, I am not sure this is urgent issue but I will look into this. So leave it open.

dprojects, 2024-01-16:
Can you add sample freeCAD file, with simple scenario, this will be easier to determine how to fix it

flashydave, 2024-01-16:
2 files. First magic_drill.FCSd contains a simple spreadsheet controlling panel size
Second magic_drill_with_holes.FCSd contains the same panel with MagicDriller used to create a series of holes on one face.
If you look at the resulting panel constraints the parameterised dimensions these are now shown with their current values replacing the previous formula. Same with any sketch. Parameterised locations are replaced with absolute values at the time of the usage of magic drill. 
[2024.tar.gz](https://github.com/dprojects/Woodworking/files/13955661/2024.tar.gz)

dprojects, 2024-01-18:
I have improved the main Cube to Pad function. So now all the conversions should keep the expressions.
Here is the video about it: https://www.youtube.com/watch?v=9_2S45Vb1-Y

Feel free to re-open this issue if there still problem related to this issue. Thanks.

closed 2024-01-18 by dprojects

flashydave, 2024-01-18:
Thanks - great response and I liked the video! 
An interesting related problem is where you might want to keep the hole locations relative to the face upon which they were initially associated so that if that face is resized in X and Y or moves in the Z axis then the holes move accordingly based on the rules that defined where they initially placed. 

Maybe there is an argument for X and Y remaining unchanged but I'm not sure you would often want the holes to stay in the same position if you change the face Z value. 

I dont know if that is possible as I suspect might you lose that upon the cube to pad construction based on what the dialogue values where at the time. However if they could retain some essence of the spreadsheet expression from the face then any small adjustments can be made with a lot less manual editing of constraints.   

Maybe something for some further development? 

Thanks