dprojects/Woodworking #33: Install of Woodworking workbench fails from FreeCAD

myhillman, 2023-12-08:
Inside FreeCAD 0.21, the Woodworking workbench is available for installation from Tools | Addon manager. The installation fails with an error message "Could not load README data from URL https://github.com/dprojects/Woodworking.git/blob/master/README.md"

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves 0.6.15
  * fasteners 0.3.50
  * fcgear 1.0.0
  * Help 1.0.3
  * parts_library
  * parts_library.bak
[/code]

jet11x, 2023-12-30:
I just installed FreeCAD and then installed Woodworking from the Addon Manager without any issues.

`
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Help 1.0.3
  * woodworking 0.21.1.33694
`

dl-home, 2024-01-08:
I tried uninstalling Version: 0.21.1.33668 +26 (Git) and reinstalling it, and I still get the error if I try to install the Woodworking workbench:    "Could not load README data from URL https://github.com/dprojects/Woodworking.git/blob/master/README.md"

Someone else mentioned that the URL is wrong.  ".../Woodworking.git/..."   should just be ".../Woodworking/..."  If that is true, is there a file I can edit on my local installation to try to fix it?

kwgagel, 2024-01-08:
I tried doing that. It fixes the description but it still can't install. Kevin W. Gagel