dprojects/Woodworking #34: Latest release workbench incompatible with latest release FreeCAD

myhillman, 2023-12-08:
As a consequence of issue #33, it was necessary to install the "Latest release" (0.20) from GitHub manually. This version is incompatible with the latest release of FreeCAD (0.21). After install the Anchor icon is missing. There may be other undiscovered incompatibilities.

I'm not sure what the answer is here, but the workbench "latest release" should be compatible with FreeCAD latest release. The install mentioned in issue #33 was attempting to install workbench 0.21, so fixing #33 will resolve this problem.

dprojects, 2023-12-09:
Latest (master) version of Woodworking workbench + FreeCAD 0.20 **IS NOT** good combination. If you want to use latest workbench version with all the features you should rather use latest FreeCAD weekly build (probably 0.22 version).

So the missing icon might be caused by the incorrect combination. However, I will look into this closer later. 

> I'm not sure what the answer is here, but the workbench "latest release" should be compatible with FreeCAD latest release. 

Your issue is correct and I like your point of view. But the Woodworking workbench master was developed before the FreeCAD 0.21 was released. So this is why I use FreeCAD Version: 0.21.30486 (Git) AppImage (at Xubuntu). This is weekly build before FreeCAD 0.21 stable.

FreeCAD 0.21.1 is not good release candidate because there is bug with TechDraw template size, it is always zero so the report can't be centered. However, maybe I will decide to release stable Woodworking version based on FreeCAD 0.21.1 with this "known issue".

myhillman, 2023-12-09:
I was not using FreeCAD 0.20. As stated, I was using FreeCAD 0.21. This is the latest stable release.

The 0.20 referred to was the Workbench version. This was offered on GitHub as the "Latest version".

As a matter of principle, the latest official versions of both FreeCAD and the workbench should be used. Other developmental and nightly builds carry risks, and they should be avoided. The latest official version of the workbench is 0.20. This needs to be upgraded.

I think the latest build for workbench needs to be updated to something that is compatible with FreeCAD 0.21. Otherwise, there will be a continuation of these compatibility issues and confusion.

dprojects, 2024-01-17:
I will try to test all feature with FreeCAD 0.21.2, and if everything will be fine I will release stable 0.21 Woodworking workbench version. But first I need to take care about the issues and fixes. Stay tuned.

closed 2024-01-17 by dprojects