dprojects/Woodworking #35: Suggestions for improving the implementation of grain

myhillman, 2023-12-11:
1. It is possible to assign all 3 grain types ("Vertical", "Horizontal", "No Grain") to a panel simultaneously. This creates a situation in the model that is impossible in real life. These 3 grain types are mutually exclusive. Assigning a grain should replace any grain already assigned.

2. The grain terminology is confusing and not the standard woodworking terminology. Consider the following scenario.

- create a vertical panel.
- Assign a vertical grain to the panel.
- A symbol showing horizontal arrows appears. This is not what the user intended.
- Now rotate the panel through 90 degrees.
- The grain marker remains the same, whereas it should have followed the orientation of the panel.

The grain terminology is confusing. "vertical" means vertical to what? It just puts vertical lines on the current face, no matter whether it is a vertically oriented panel or a horizontally oriented panel. When the panel is rotated, the grain symbol remains unchanged.

It would be be better if standard woodworking terminology was adopted. A grain can be either "straight" (i.e. longitudinal) or "cross". [https://en.wikipedia.org/wiki/Wood_grain] Arguably there is also "end" grain. I would suggest the following changes.
a. There are 4 choices of grain - straight, cross, end and none.
b. A panel can only be one grain at a time.
c. When assigning a grain (straight or cross), the grain direction is defined by the longest (straight) or shortest (cross) dimension. I don't know what to do if the panel is square.
d. The grain arrows are rotated if the panel is rotated, and maintain their alignment to the longest/shortest side.

dprojects, 2024-01-16:
1. I'm more of an advocate of adding options rather than banning them. However, I agree that if grain were replaced with another one, it would be faster than additional deletion.
2. I need to figure it out, more deeply later to understand it better.

myhillman, 2024-01-16:
I don't understand your comments at all. I am not advocating adding or deleting anything. It's a simple matter of the grain feature not representing real life. When a panel with grain specified is rotated, the grain should also rotated. It's as s simple as that. A panel can also have only one grain at a time.

dprojects, 2024-01-16:
> When a panel with grain specified is rotated, the grain should also rotated.

Generally rotation is complicated. If you use a lot of rotation, you should rather put your elements inside container and rotate container not single elements. 

Single element rotation is much more complicated feature and is not implemented yet. For example if you have rotated panel, it is rather hard to determine the exact place for dowels, you need to calculate the rotation, it is not so simple and not so precised. Also rotated elements not have its plane XYZ so they might be very problematic. 

However, the grain rotation is probably simple thing to do and probably will be done. Stay tuned.

dprojects, 2024-01-17:
Hello again, I have just improved the grain direction a little bit.
Now new grain direction description replace the old one.
Also you can apply the grain direction for rotated Cube (without container).

Here is video for that: https://www.youtube.com/watch?v=D9LWHyqF4rg

Note: Pads are not supported. This is because Pads are little more complicated. You can rotate Pad via Sketch and also via Body. But you can have many nesting Body objects. So it is little implementation problem to zero and sum up and back all the nesting Body objects... probably this would be possible to do but I would rather postpone it until rotation feature to be completed. 

Let me know how urgent and useful are Pads for you.
Feel free to reopen this issue or discuss here. Thanks.

closed 2024-01-17 by dprojects