dprojects/Woodworking #36: Addon Manager entry is broken due to a typo

BurntVoxel, 2023-12-11:
... does this bug belong to you guys or to the FreeCAD addon repository? I can't tell. 
Anyway the problem is that FreeCAD's addon manager can't fetch this plugin, because the link has a typo. The listing is visible, but when you go to view the details or install it, it 404s.

Correct: <https://github.com/dprojects/Woodworking>
Current: <https://github.com/dprojects/Woodworking.git>

[image]

kwgagel, 2024-01-01:
I am seeing this issue as well.

kwgagel, 2024-01-14:
I made a workaround to get it installed.

1. Open FreeCAD
2. Open the AddOnManager
3. Click Refresh local cache and wait for it to completely download a new cache
4. Open File Explorer and navigate to: c:\Users\**YOURUSERACCOUNT**\AppData\Local\Temp\FreeCAD\Cache\AddonManager\PackageMetadata\woodworking
5. Open the 'package.xml' file with Notepad
6. Edit the '<url type="repository" branch="master">https://github.com/dprojects/Woodworking**.git**</url>' line by removing the **.git** so it looks like this '<url type="repository" branch="master">https://github.com/dprojects/Woodworking</url>'
7. Save the package.xml file
8. Close the Addon Manager
9. Open the Addon Manager again and wait for it to finish reading the cache
10. Search for Woodworking
11. Try the install again

dprojects, 2024-01-16:
check if there is still problem with last master branch

closed 2024-01-16 by dprojects