dprojects/Woodworking #38: Feature: Quick constraints

garthwilliamson, 2024-01-07:
Hi! I have a set of macros I've been using now I discovered this workbench (brilliant work btw!) and I don't know if they are useful more generally. They're for contraining boxes relative to each other making it much quicker to build paremetric models. I have been using them for creating frame + skin designs for absolutely-not-fine-woodworking projects, I'm still working out if they can be extended to more complex joint types. I create a spreadsheet with the driving dimensions and then place everything relative to those or other objects.

I'm not offended if this kind of idea isn't useful to more than just me, but it _might_ be something worth adding.

This only works with objects that have length, height and widths. I'm still trying to learn a workflow to use the boxes as the base objects and then layer the detailed joints on top of them.  I create 3 different macros so I can hotkey creating these constraints:

```python
import FreeCADGui

choice = "x"

direction_index = {
    "x": 0,
    "y": 1,
    "z": 2
}[choice]

wordy_direction = {
    "x": "Length",
    "y": "Width",
    "z": "Height",
}

thing_a, thing_b = FreeCADGui.Selection.getSelection()

ex = FreeCADGui.Selection.getSelectionEx()

face_a = ex[0].SubObjects[0]
face_b = ex[1].SubObjects[0]

offset_a = face_a.Surface.Position - thing_a.Placement.Base

offset_b = face_b.Surface.Position - thing_b.Placement.Base

# If this is non-zero, take the component in the direction_index direction and
# make sure it matches lenth/width/height as appropriate.

formula = f"<<{ thing_a.Label }>>.Placement.Base.{ choice }"

if offset_a[direction_index] != 0:
    formula += f" + <<{ thing_a.Label }>>.{ wordy_direction[choice ] }"

if offset_b[direction_index] != 0:
    formula += f"- .{ wordy_direction[choice] }"

print(formula)
thing_b.setExpression(f".Placement.Base.{ choice }", formula)
thing_b.recompute()
```

dprojects, 2024-01-16:
I am not sure what is your goal, and how I can help you?

dprojects, 2024-01-16:
> if this kind of macro is generally useful to you

I am not into adding someone's code because this is not community project. This is more private project with sharing for others. 

Adding someone's code and managing community projects with teams cause many problems, is more hard. There are problems with testing, make it compliant with coding standards, compliance of licenses and copyright resolutions.

However, I open for professional knowledge sharing and issues. So, if you want feature or any help with code maybe I can help.

garthwilliamson, 2024-01-22:
Thanks for the offer, I'll work on my own set of macros and plugins and maybe release them later separately, if I can't find someone else who has done the same thing!

closed 2024-01-22 by garthwilliamson