dprojects/Woodworking #41: Rules for the Wiki FC to Docusaurus converter.

kaktusus, 2024-02-20:
# Rules for the Wiki FC → Docusaurus converter.
Working name: **_Wiki2Docu converter_**

### Conversion tools.

Conversion tools are created on the fly. More information can be found on [kaktus' GitHub](https://github.com/kaktusus/FreeCAD-translators-tools/tree/main/Wiki2Docu).

### Consider whether transferring the functionality of "Wiki templates" is helpful and feasible for implementation.

If the decision is made to transfer this functionality, the first step is to prepare such templates.

### Define conversion rules for the used Wiki templates.

This will allow the application of appropriate algorithms during the preparation of the converter, which will generate the required content of the generated files.
The more that is implemented, the less work there will be for editors.

The created templates are available on the [WikiPages, Templates section](https://wiki.freecad.org/WikiPages#Templates).
Defining the rules to be applied during conversion will facilitate the definition of converter algorithms.
#### Rules for the used Wiki templates.

1. FreeCAD Wiki internal links `[[WikiPages|Wiki Pages]]`,`[[WikiPages/pl|Strona Wiki FreeCAD]]`
    Define rules for the location of documentation files to automatically insert correct links during conversion.
2. Tool icons and others with a defined size for display, e.g., `[[File:PartDesign_Body.svg|16px]]`
    Define the locations of graphic files to automatically insert correct links during conversion.
    ***Suggestion***: the Wiki contains one main directory for graphic elements.

dprojects, 2024-02-21:
What do you mean by that? Should I create wiki page? update README.md files or what should I do with it? ?

kaktusus, 2024-02-21:
cześć Darek. :star_struck:

As the maintainer of a cool workbench, you are not obligated to take any actions. This is a separate topic directly related to migrating documentation to a new platform. 
In order for me to develop a comprehensive tool that will facilitate painless generation of documents in the new format, I need a few guidelines.

If several people participated in the discussion, it would be excellent. The results could be better.

I just realized that I started this topic in the wrong place. :rofl:

By the way, would you agree to allow translations of your workshop to be imported into Crowdin?

dprojects, 2024-02-21:
> I just realized that I started this topic in the wrong place. 🤣

No problem with that.

> By the way, would you agree to allow translations of your workshop to be imported into Crowdin?

I guess, anyone is allowed to make its own translation and add it to Crowdin according to MIT license? 
I should do something?

I should keep this issue open here?

kaktusus, 2024-02-22:
Crowdin is essentially a platform for community-driven translations. Collaborative work yields better results. 
The entire international FreeCAD community has access to the prepared  strings.

The basic workflow is as follows:
- Extraction of strings after each code update,
- Uploading the .ts file to Crowdin,
- Preparation of translations,
- Importing translations into the source code.

In this thread there is additional information on the subject.
https://github.com/shaise/FreeCAD_FastenersWB/issues/340
If you agree I will take care of the eteps related to the handling of Crowdin. :blush:
Besides, this topic can be closed.  :upside_down_face:

kaktusus, 2024-02-28:
So in short, you don't mind bringing translations of your workbench to Crowdin? 
And do you agree?

dprojects, 2024-02-28:
> you don't mind bringing translations of your workbench to Crowdin

I don't know what you mean by bringing translations to Crowdin... anyone can do that.
I am not into giving administration permissions of my github account to any application... this should be manually reviewed, approved and submitted, in my opinion.

kaktusus, 2024-02-28:
Can I upload the extraction string file to Crowdin?

Albo tak:
Darek czy nie masz nic przeciko abym załadował plik ze stringami do tłumaczenia na platformę Crowdin.
Nie są wymagane żadne połaczenia z twoim kontem GitHub.
Wybacz za mój angielski.
:cactus:

dprojects, 2024-02-28:
> Darek czy nie masz nic przeciko abym załadował plik ze stringami do tłumaczenia na platformę Crowdin. 

Przecież licencja [MIT](https://github.com/dprojects/Woodworking?tab=MIT-1-ov-file#readme) jest najbardziej liberalna ze wszystkich i pozwala nawet na modyfikację kodu, a wygenerowane frazy do tłumaczeń to nawet nie jest kod, tylko już twój własny produkt. Tak przynajmniej ja to rozumiem. Więc możesz chyba z tym zrobić co chcesz.

> Nie są wymagane żadne połaczenia z twoim kontem GitHub.

Tylko cały problem polega na tym że jak zrobię jakąś zmianę kodu to już może nie działać bo tam chyba jest odniesienie do numeru linii w kodzie. Nie wiem jak to działa bo tego nie testowałem. 

A jak byś chciał żeby tłumaczenie znalazło się w masterze to trzeba zrobić submit, wtedy bym musiał twój plik z tłumaczeniem wrzucić do mastera ale najpierw przetestować czy to działa...

kaktusus, 2024-02-28:
I owszem. Ale z grzeczności zapytałem czy wyrażasz zgodę na wdrożenie tłumaczeń na Crowdin. Potraktuj to jako prestiż.
Powiesz że nie chcesz nic nie zrobię w tym zakresie.

Tu też masz rację. Każda zmiana kodu dotycząca stringów do tłumaczeń wymaga kolejnej akcji. Czyli ekstrakcji stringów i załadowania do Crowdin aktualizacji.
Potem tłumacze pracują ...
Kolejnie co jakiś czas warto zrobić PR z wyeksportowanymi stringami z Crowdin. Zatwierdzasz wtedy zmiany i tyle.

Ponieważ nie masz konta na platformie Crowdin mogę Cię wspierać w tym zakresie.
Podstawowy przepływ pracy:

- kaktus ładuje plik źródłowy na Crowdin.
- tłumacze pracują.
- kaktus eksportuje tłumaczenia i robi PR.
- Darek zatwierdza aktualizacje tłumaczeń.
- Darek rozwija kod, wprowadza nowe funkcjonalności, które wymagają nowego importu do Crowdin. Informuje kaktusa.
- i tak dalej.

Jeśli kiedyś byś stwierdził że chcesz poznać funkcjonalność platformy Crowdin w omawianym zakresie, wystarczy, że dasz mi znać. :wink: 

Oczywiście co jakiś czas, warto wykonać aktualizację tłumaczeń - w miarę postępu tłumaczeń różnych narodowości.

I oto cały ambaras.

dprojects, 2024-02-29:
> czy wyrażasz zgodę na wdrożenie tłumaczeń na Crowdin

To miłe że pytasz, masz racje zawsze bardziej uprzejmie jest zapytać. Pewnie że możesz wrzucić tłumaczenia na Crowdin.

> Ponieważ nie masz konta na platformie Crowdin

Mam konto tutaj [Darek at Crowdin](https://crowdin.com/profile/dprojects)
Kiedyś założyłem tam konto ale doszedłem do wniosku że w sumie to tłumaczenie można zrobić w dowolnym edytorze tekstowym. Nie zajmowałem się tłumaczeniami więc nic tam nie robiłem i chyba tam trzeba coś wykupić żeby działać.
Jedyna zaleta Crowdin jaką widze to międzynarodowa społeczność która może pozwolić na robienie tłumaczeń w innych językach których się nie zna. To może być faktycznie użyteczne narzędzie z tej perspektywy, pytanie tylko jak z jakością takich tłumaczeń...

>     * kaktus ładuje plik źródłowy na Crowdin.
>     * tłumacze pracują.
>     * kaktus eksportuje tłumaczenia i robi PR.
>     * Darek zatwierdza aktualizacje tłumaczeń.
>     * Darek rozwija kod, wprowadza nowe funkcjonalności, które wymagają nowego importu do Crowdin. Informuje kaktusa.
>     * i tak dalej.
> 
> Oczywiście co jakiś czas, warto wykonać aktualizację tłumaczeń - w miarę postępu tłumaczeń różnych narodowości.

Ja bym proponował robić tłumaczenia tylko dla [Releasów](https://github.com/dprojects/Woodworking/releases) bo tam ten kod już się nie zmieni. Wtedy tłumacze mogą sobie pracować w oparciu o stały niezmienny plik z frazami w dowolnym czasie. Nie wiem czy github daje taką możliwość do dorzycenia czegoś do releasu, jeśli tak to ewentualnie można by było tam dorzucić tłumaczenia. A jak nie nie to w sumie mogą też być dostępne na Crowdin, nie wiedzę problemu.

Jak chcesz robić tłumaczenia dla mastera to wtedy by trzeba było wyznaczyć jakiś czas na zrobienie tłumaczeń przed wypuszczeniem releasa żeby to wszystko zgrać. Ale wtedy tłumacze są ograniczeni czasowo. Moim zdaniem lepiej jest robić tłumaczenia już dla zamkniętego kodu który się nie zmieni.

kaktusus, 2024-02-29:
No i super. :smiley: 

Skoro masz też konto poproszę o podniesienie Ci rangi do "developer", abyś mógł sobie pozaglądać od środka jak to działa.

Poruszyłeś też istotny temat, na który Crowdin też jest przygotowany. Zobacz: https://support.crowdin.com/versions-management/

Cała reszta do ogarnięcia i umówienia się co i jak...

dprojects, 2024-02-29:
OK, to w takim razie zamykam ten temat. Jakbyś już miał tłumaczenia do wersji stabilnej to daj znać w tym wątku albo innym, wtedy zobaczę czy da się dorzuć to do releasa...

closed 2024-02-29 by dprojects