dprojects/Woodworking #43: documentation

aarmehl, 2024-02-27:
Hi,
The English documentation is very hard to read. While I am willing to fix it a little bit, I don't have the time to actually to write the documentation. (I was a technical writer once upon a time) 
 I also never used any CAD programs before and I  and I don't see any definition of terms. Youe youtube demos have no talking explaining what you are doing, so they offer me limited help. 
I'm,sure that if I had some background using CAD programs I would probably understand where to start.  I want to design a workbench using freecad and your workbench, but I have no idea how do it with freecad and your workbench. Also how do I set up the woodworking workbench to use inches and feet instead of metric.
Are you help files written in HTML or Markdown? thanks, 
Aaron 
mehlzaidy770@gmail.com

dprojects, 2024-02-27:
> Hi, The English documentation is very hard to read. 

Sorry, I am not native English speaker.

> While I am willing to fix it a little bit, I don't have the time to actually to write the documentation. 

No problem, this is a hobby project, so the time spent on it is limited.

> I have no idea how do it with freecad and your workbench. Also how do I set up the woodworking workbench to use inches and feet instead of metric. 

This is related to FreeCAD documentation, you can set it in FreeCAD settings, not in workbench settings.

> written in HTML or Markdown?

Markdown

> youtube demos have no talking explaining what you are doing, so they offer me limited help

This is a hobby project, it is not sponsored by any large corporation, so the documentation is not at a sufficiently high level. I have neither the time nor the opportunity to prepare professional training or conferences on the use of this environment. Perhaps someday, when someone becomes seriously interested in it, maybe something like this will be created, but for now it is as it is.

aarmehl, 2024-02-28:
I set up freecad for inches but wood working still liked metric.

aarmehl, 2024-02-28:
I downloaded the documents but it tells me its a readme only file. How do I
get one I can edit?
Thanks,
Aaron

dprojects, 2024-02-28:
> I downloaded the documents but it tells me its a readme only file. How do I get one I can edit?

The README.md files contain the Markdown text, so you can edit them.

> I set up freecad for inches but wood working still liked metric.

I guess, you mean the default panels, tooltips and some pop-ups... they are hard-coded and they are not recalculated to inches... so this might be the issue...
Let me think if this can be changed and this is worth to change...

aarmehl, 2024-02-28:
I converted the markdown to Dita. I did some edits today. How do I quickly
resize a panel? Is there a way to do it with a mouse?
Making it possible to use inches and feet would help since that's how
lumber is sold here.
Thanks,
Aaron

dprojects, 2024-02-28:
> How do I quickly resize a panel? Is there a way to do it with a mouse? 

You can use [Resize panels icons](https://github.com/dprojects/Woodworking/tree/master/Docs#resize-panels)

> Making it possible to use inches and feet would help since that's how lumber is sold here.

Probably the problem with metrics is only with [Default panels](https://github.com/dprojects/Woodworking/tree/master/Docs#default-panels), tooltips, and popups. All features not depends from units.

aarmehl, 2024-03-05:
HI again,
could you simply explain what copy panel does. The help file is confusing
me.
Thanks

dprojects, 2024-03-05:
> could you simply explain what copy panel does. 

For example if you have panel with XY orientation and you want new panel with the same dimensions but with XZ orientation, you can quickly select the panel XY and click the copy XZ icon. 

But more useful feature is if you have Pad or any other PartDesign object and you want to have Cube panel with the same dimensions, so you can select the PartDesign object and click exact copy icon and you have panel with the same dimensions but Cube, so this is some kind of Pad -> Cube transformation... 

> The help file is confusing me

Do you have any idea how to improve the description?

aarmehl, 2024-03-05:
I'm working on it.
I don't think so well in the axis orientations.
top=xz?
front=
side view=?
I once I can write a  it using simple language, I can put the axis
coordinates in parentheses.
I'm still new to CAD. Also I don't find a Show Constraints icon, is it part
of another workbench??
Thanks
Aaron

dprojects, 2024-03-06:
> I don't think so well in the axis orientations. top=xz? front= side view=? 

I guess, no, you have to see the right bottom corner at 3D view, there are XYZ axes, so the panel should rather be oriented according to it.

> I once I can write a it using simple language, I can put the axis coordinates in parentheses. I'm still new to CAD. 

I don't know what you mean by "axis coordinates in parentheses". I didn't use any CAD program before FreeCAD as well. 

> I don't find a Show Constraints icon, is it part of another workbench??

Constraints are used at Sketcher workbench, mostly for PartDesign objects. But Woodworking workbench recommend little different approach. You should rather create full model from Cubes (panels) not bounded with each other. So you will be able to move and replace any part later. At PartDesign you build one object at another, so it is slower to recalculate full model and in many cases you can't remove the bottom parts.

aarmehl, 2024-03-06:
Wait, your docs for the magicresizer say specifically if the object has no
constraints at the selected edge they won't be resized. Use the sho
constraints tool to check it.

That said, I can't figure out a work flow from parts to a complete design.
I'm designing a work bench.
It has a
plywood top of US 1.5x24x48
4 legs US 4x4
6 beams 2x4x18
6 beams 2x4x21
2 plywood shelves 20x40x.25
All in US inches.
What tools do I use to design it?
Is there an easy to create the2x4's?

Knowing how each tool works doesn't translate into knowing how to use the
workbench
.
Thanks
Aaron

aarmehl, 2024-03-06:
Here is an example of a workbench I got off of the net.

aarmehl, 2024-03-10:
> Yes but this is only for Pad (PartDesign objects based on Sketch), if you
> have Cube they are easier to manage and this will be resize.
> No matter what you want to design you can build it from separate parts
> (panels) just as children build more complex objects with Lego bricks and
> just as furniture is created in the real world.

So I now can build one table leg.
But I don't see how to use it to make 3 more
I can make a table top but I have no idea how to connect the parts.
Could you send me the actual file for the table you made so I can learn
from it.
Is there a way to list all the parts I want in a spreadsheet, and then have
freecad make them?
I need somewhere that doesn't explains tools but explains how to make a
piece  of furniture.
Thanks,
Aaron

dprojects, 2024-03-10:
> So I now can build one table leg. But I don't see how to use it to make 3 more

You can use PartDesign Mirror but I would rather recommend using magicMove to copy...

 > I can make a table top but I have no idea how to connect the parts

You don't have to connect parts...

> Could you send me the actual file for the table you made so I can learn from it

See the first movie at youtube... as I remember there is table...

> Is there a way to list all the parts I want in a spreadsheet, and then have freecad make them

No but you can make table parametric via spreadsheet...

>  I need somewhere that doesn't explains tools but explains how to make a piece of furniture

You have many videos at youtube...

aarmehl, 2024-03-11:
I found the YouTube videos, and I made my first table.

dprojects, 2024-03-11:
> I found the YouTube videos, and I made my first table.

Great to hear it ! Good Luck !

but I keep this issue open because I need to figure out the units... 
the bounding feature (to connect pieces on demand) will be implemented when I will be in the right mood to do it...

aarmehl, 2024-03-14:
Also I saw the mortise and tenon video, but I wasn't sure I saw one on
making the other joint types. could you steer me in the right direction?
Thanks,
Aaron

dprojects, 2024-03-14:
> Also I saw the mortise and tenon video, but I wasn't sure I saw one on making the other joint types. could you steer me in the right direction? 

The youtube videos are not step by step tutorials... you need to practice on your own and rely on your own creativity. However you may want to watch the: [joinery playlist](https://www.youtube.com/playlist?list=PLSKOS_LK45BBG8kJ2AZvQKBfOSfzhTrLt)

closed 2024-12-04 by dprojects