dprojects/Woodworking #44: magicDriller not working in Freecad v0.22

blob42, 2024-04-29:
Hi, 

Thanks for your efforts with this workbench. I am using freecad 0.22 and the magicDrill does not work as intended. The drill bits appear and move as expected but when I press on `drill below drill bits` a new Part is created with an empty sketch named `Pattern, Panel XX` See the screenshot below

[image]

blob42, 2024-04-29:
Sorry It works I just realized I was using it the wrong way

dprojects, 2024-12-04:
this is still an issue? I can't reproduce this problem... please reopen this issue if there is still something unsolved.

closed 2024-12-04 by dprojects