dprojects/Woodworking #46: Golden Rules - Examples for clarifications and why for someone new?

numericOverflow, 2024-09-29:
First off, this workbench is AMAZING!  I was struggling for quite a while to do some CAD for a project using wood and using this workbench, I'm finally making it happen.  Kudos for all the hard work!  Your existing documentation is extremely helpful to quickly understand most of the tools in the workbench.

Would you be able to make a video(s), or perhaps explain the reasons behind your "Golden Rules"?  Specifically the first couple rules talking about don't rotate/pad/etc directly, use LinkGroups & Containers.

> Golden rules:
> - Don't rotate objects directly, rotate them via container LinkGroup.
> - Don't copy Pad directly. Copy, Clone, Link the Part container.
> - Don't mix Cut with PartDesign too much. Keep clear and simple design line.
> - If you want generate cut-list, BOM, dimensions, rather avoid packing objects extremely, for example Array on Array or MultiTransform on MultiTransform.
> - Rather not move objects via AttachmentOffset, move them via container Body, LinkGroup.
> - Design furniture from Cubes. If you want more detailed model convert exact element to Pad and edit the Sketch. Also for irregular or not rectangle shapes.

Right now, I've been creating panels, then rotating them using your MagicMove & MagicAngle tool.  That seems like the right workflow, but once I saw your Golden Rules called out using LinkGroups & Containers, I started wondering if I'm doing it wrong. 

I tried looking around the README and examples, but didn't find much of anything with specifics about the rules and why you came up with them.  Additional details or a video explainer would be very helpful to get up to speed.

dprojects, 2024-09-30:
The Golden rules are overall approach to avoid some issues. For example if you rotate Pad or Cube directly and you want to apply dowels via magicDowels tool, this will not be working, because the XYZ plane has been changed. But if you put Cube or Pad inside LinkGroup container and rotate the whole LinkGroup container you will be able to apply dowels via magicDowel tool because the XYZ plane will not be changed.

But if you don't want to use magicDowel or any other tool that use XYZ plane, everything should be fine. You can always later zeroed the rotation and put the "problematic" element inside the container and rotate the LinkGroup container. So even if you do such "mistake" you can repair it later.

dprojects, 2024-12-04:
Please reopen this issue if there is still something unsolved.

closed 2024-12-04 by dprojects