dprojects/Woodworking #49: Support for FreeCAD 1.0 and later

dprojects, 2024-12-16:
Currently there are no plans to support FreeCAD 1.0 and later. 

The main reason for that is that FreeCAD 1.0 no longer allow for direct Sketch move, the [FreeCAD team consider direct Sketch moving as bug](https://forum.freecad.org/viewtopic.php?p=794690#p794690), so they fixed this "bug"... 

Using FreeCAD 1.0 with Woodworking workbench for example you can't use [magicMove](https://github.com/dprojects/Woodworking/tree/master/Docs#magicmove) to move Sketch or Pad. Also you can't move holes after drilling or center holes or Pads. So this "fix" impact many advanced features at Woodworking workbench.

However all these feature works fine with FreeCAD 0.21.2 and personally I don't see any reason to use FreeCAD 1.0. 
Also I am happy with FreeCAD 0.21.2 and I don't see any reason to chase all the FreeCAD team changes and "fixes" and adjust Woodworking workbench code only to have new FreeCAD number...

The Woodworking workbench now has its own [release versioning system](https://github.com/dprojects/Woodworking/releases), the 0.22 release has been published, and will be developed regardless of FreeCAD politics.

SnowBeaver, 2025-01-09:
Sorry to hear that. However, I have been using magicmMove with version 1.0 without problem. Anyway, I was writing about a bug with magicMeasure, as the tool do not measure a all and keeps generating Distance element in the TreeView. 

If magicMove works with Freecad 1.0, will you consider supporting it?

Michel Belanger, Canada

dprojects, 2025-01-10:
> Anyway, I was writing about a bug with magicMeasure, as the tool do not measure a all and keeps generating Distance element in the TreeView.

I didn't test all features so carefully because I saw I can't move Sketch at FreeCAD 1.0 what makes the FreeCAD 1.0 useless for me. Please open new issue and describe it, maybe this bug is also at FreeCAD 0.21.2.

> However, I have been using magicmMove with version 1.0 without problem.

At FreeCAD 1.0 you can't move Sketch by selecting it, or move Pad by selecting Pad. You can only move Sketch or Pad inside container, for example by selecting Body. But the bigger problem is their politics, they consider Sketch move as bug, so probably there will not be setGlobalPlacement() function for Sketch object.

> If magicMove works with Freecad 1.0, will you consider supporting it?

If someone find it working, it is fine to me. But I can't recommend FreeCAD 1.0.
FreeCAD is released with AppImage so you can have more than one version of FreeCAD at your system. 

> Sorry to hear that.

Yes it is little sad but I guess the Woodworking workbench should be rather considered as new different product dedicated for woodworking but based on FreeCAD kernel. Woodworking workbench has too much features to be always up-to-date with any FreeCAD changes. 

But maybe someday the FreeCAD kernel will be upgraded from 0.21.2 into something above 1.0. 
Currently I want to focus on make the Woodworking workbench stable with FreeCAD 0.21.2 kernel and create maybe some more useful features. I think chasing all FreeCAD changes is not good approach.

SnowBeaver, 2025-01-10:
I understand your position, Woodworking is a tool for yourself and it does not really matter which Freecad version you are using.

As for the problem using magicMeasure, it is easy to reproduce in FC 1.0 as it does not work at all. I simply click on the tool and the mearements keep showing 0,000.

Michel

dprojects, 2025-01-10:
> Woodworking is a tool for yourself and it does not really matter which Freecad version you are using.

It is natural process, more features you have, more problematic is to keep all the features in sync with other dependent software, especially if they don't care about backward compatibility. So maybe it is time to go with my own way...

> As for the problem using magicMeasure, it is easy to reproduce in FC 1.0 as it does not work at all. I simply click on the tool and the mearements keep showing 0,000.

Currently I am going to add new feature "Mirror Copy" to magicMove to be able to copy part of the furniture with Edge selection... Also I want to add possibility to create drawers in series. 

However, I don't say no to FreeCAD 1.0 fixes. They change `Sketch.Support` into `Sketch.AttachmentSupport` so the Cube ->Pad conversion wasn't possible at FreeCAD 1.0 but [I made small fix](https://github.com/dprojects/Woodworking/commit/296131990e7a1c12bd02a2d1e4a378bbd99ab5d7) and now this working in FreeCAD 0.21.2 and also in FreeCAD 1.0. If the same will be possible with `magicMeasure` I will do it.

SnowBeaver, 2025-01-11:
I started using your workbench once FC 1.0 was released. Before FC 1.0, FC was too buggy and I did not bothered to seriously try workbenches. I imagine other users will do the same. The video you made help a lot. The doc itself is harder to follow. Anyway, for me the video made a difference. Excelent tool overall.

I am building a cabinet (3 meters x 2 meters) for my daughter, and one new feature I would appreciate is the ability to copy a cabinet section and change one of its dimension. At the moment, I can copy a cabinet section, but to change the cabinet dimension requires to modify each panel one by one. May be it can be already done with one of your tools but reading through the docs, that function did not appear to be available?

I tried to update your workbench but nothing was avalable. Your update have not been publish yet I assume?

 Michel

dprojects, 2025-01-11:
> I started using your workbench once FC 1.0 was released. Before FC 1.0, FC was too buggy and I did not bothered to seriously try workbenches. 

There is no code without bug. Bug can be opportunity for nice useful feature, like for example direct Sketch move. 

> I imagine other users will do the same. 

If they only looking for FreeCAD version number, not for doing something useful, probably yes.

> I am building a cabinet (3 meters x 2 meters) for my daughter, and one new feature I would appreciate is the ability to copy a cabinet section and change one of its dimension. 

You can share your project file at [FreeCAD forum](https://forum.freecad.org/viewtopic.php?f=3&t=8247) so maybe I will be to look at it and maybe improve the Woodworking workbench to make it more easier to do. This issue is related only for FreeCAD 1.0. However you can also open new issue and share your file there.

> to change the cabinet dimension requires to modify each panel one by one

It is FreeCAD feature, yOu can hold left-CTRL to select more panels and change the dimensions in FreeCAD Data window. You don't need special tool for that.

> I tried to update your workbench but nothing was avalable. Your update have not been publish yet I assume?

Currently I am during magicMove improve, it is not submitted yet. I don't know when it will be available.

dprojects, 2025-01-19:
> As for the problem using magicMeasure, it is easy to reproduce in FC 1.0 as it does not work at all. 

Because they changed measurement objects fields from `.P1` into `.Position1` and the `.P2` into `Position2`. I have already add [support measurement for FreeCAD 1.0](https://github.com/dprojects/Woodworking/commit/10c971b95c201f4e0fc839aceea0392ee622c791) and now it is working in `hover off` mode.

> I simply click on the tool and the mearements keep showing 0,000.

However, you are the lucky one because for me in `hover on` mode the FreeCAD 1.0 crashes, probably adding and removing objects in Tree is changed as well... as I mentioned before, they also changed `Sketch.Support` to `Sketch.AttachmentSupport`.

Don't get me wrong, but I don't recommend anyone to use FreeCAD 1.0... it's really ridiculous, it seems like they've never heard of backward compatibility...

SnowBeaver, 2025-01-19:
I can understand the despair feeling seeing your work ruined by
unconsidered development rules.

Where can I get your recommended FC version?

dprojects, 2025-01-19:
> I can understand the despair feeling seeing your work ruined by
> unconsidered development rules.

The ruined many workbenches and macros for sure, not only my work, by creating new objects fields without linking the old names to the new one. But looks like they don't care about it. It is little strange because why they allow for python scripting and making macros and workbenches if they "just changing" fields names? I don't understand their politics... sounds like joke to me...

> Where can I get your recommended FC version?

It is described at main page... [FreeCAD 0.21.2](https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.2) but maybe I should release my own product because you never know what they have in mind and who paid them...

txtinman, 2025-02-15:
I use FC only for woodworking projects. The woodworking workbench makes it much easier for me to design my projects. I don't care what version of FC you use. If you create a separate application then I will use it.

dprojects, 2025-02-16:
> I use FC only for woodworking projects. The woodworking workbench makes it much easier for me to design my projects. I don't care what version of FC you use. If you create a separate application then I will use it.

It's no problem to have version AppImage 0.21.2 for Woodworking workbench purposes and version AppImage weekly for other purposes. I personally have several versions of AppImage under the Xubuntu operating system and there are no problems with it. However, I don't see anything interesting in the new version of FreeCAD 1.1 that would somehow enhance furniture creation.

Support for higher versions of FreeCAD 1.1+ will be considered, but first I would like to do some more functionalities that are useful in practice and improve the documentation a bit, so that it is all at a stage that I consider cool and worthy of a new stable version. Then I will release another version 0.23 based on the FreeCAD 0.21.2 kernel and then I will start thinking about support for newer versions of FreeCAD.

However, the problem of supporting FreeCAD 1.0 has highlighted the problem of the FreeCAD team's politics that has been there for a long time since I started creating any macros. I don't want to go into details here and accuse anyone... but sometimes I have the impression that the FreeCAD team feels great injustice and even hatred towards anyone who creates some valuable macro or workbench just on the basis of "because it's not ours"...

That's why I don't know if it wouldn't be better to completely cut myself off from these people... and run my own project... I'm still thinking about it, but for now I'm doing what I like the most, which is programming, creating new functionalities, inventing new solutions, making work easier for myself and others...

Perhaps it would be better if Woodworking workbench attracted only people who are looking to do something practical in the real world, design and make some furniture, than those who install Woodworking workbench just for the version number...

BTW: I used Google Translate to make it faster, because I have to go and finish a new feature...

awschult002, 2025-04-21:
I don't know if this falls into the bucket of 1.0 support or not; but for my workflow, I really really need assemblies. Is there anyway to get the containers for pre-made furniture into a freeCad Assembly structure? This would allow super fast work of TechDraw blueprints and exploded views.

dprojects, 2025-04-21:
> I don't know if this falls into the bucket of 1.0 support or not; but for my workflow, I really really need assemblies. Is there anyway to get the containers for pre-made furniture into a freeCad Assembly structure? This would allow super fast work of TechDraw blueprints and exploded views.

This is not related to supporting a new version of the FreeCAD kernel, just new functionality. So, I created new issue and I will answer there.

closed 2025-06-03 by dprojects