dprojects/Woodworking #50: Question: How-to undo?

connde, 2025-01-19:
Hi, first off all, thanks for developing this add-on, is a create toolset.

I'm going through some tutorials and I found that all the actions I do cannot be undone, only the native actions from FreeCAD, is that right? Is there a way to undo an action? For example, when you use magic angle or magic move?

Using FreeCAD version 0.21.2.

Thanks in advance.

dprojects, 2025-01-20:
> I'm going through some tutorials and I found that all the actions I do cannot be undone, only the native actions from FreeCAD, is that right? Is there a way to undo an action? For example, when you use magic angle or magic move?

The magicAngle is easy to undo because there is angle step, so if you choose the same rotation point it will be in the same place, if... but if not it is better, and quicker, to remove the object and create new one... :-)

Yes, it's true, there are many situations that can cause frustration, that something broke after "incorrect" clicking. For me personally, the most annoying thing is moving, positioning many objects, mainly PartDesign type. Often some objects fall apart, which can be annoying.

FreeCAD has a certain mechanism for undoing changes, but it doesn't always work. There is also a certain mechanism that can be used when creating code, which in a sense records script changes, but I personally don't use it in the code, maybe I should take it more seriously, but on the one hand I also doubt that it will be possible to undo everything. For example, if you change a Cube type object to a Pad, let's say while drilling in a Cube type object, it is automatically changed to a PartDesign type object, and additionally the Cube type object is deleted, so it will probably be hard to restore everything... although I don't know.

For now, I will leave this issue open, but there is little chance that it will be solved quickly, although I know that it is an important issue. The best way is to make backups of the project and get more practice using some tools, especially positioning tools, because positioning in FreeCAD is generally a difficult subject, because an object doesn't really know its global position, like it does in containers... so positioning in a sense wasn't solved by FreeCAD.

The best example is when you have a Pad object, offset to AttachmentOffset, but when you create a Clone, the Clone will have position (0,0,0) and will not be in position zero, additionally if you throw it into many containers and make offsets on the containers, no FreeCAD function will return the correct position of such an object. This is a bit off-topic.

dprojects, 2025-02-26:
Hello again after long time, I just added [support for undo things](https://github.com/dprojects/Woodworking/commit/d83d8678cc3414126aba6f5bfd6bb3d8eaa9c22a), not much code changes but for sure significant change from user experience perspective.

Of course, this doesn't solve all the problems with undoing changes, but the more annoying ones, especially regarding moving and rotating objects.

However, I noticed that FreeCAD undoes all changes from a given file, it doesn't seem to queue them relative to objects or the order of operations from a given file. I don't know if there's also some list in FreeCAD where the changes are and you can, for example, undo a change from the middle of the list... I haven't delved into it that much.

However, all these changes should help a lot, and it will certainly be less annoying, especially when someone's new is learning how to use the tools.

If you see anything else that's annoying that's worth fixing, let me know.

closed 2025-02-26 by dprojects