dprojects/Woodworking #51: magicStart Furniture: No cabinet available with faceframe

awschult002, 2025-01-27:
I absolutely love this package and am super excited to be using it for my wardrobe and custom cabinet projects. I have noticed that the magicStart furniture list doesn't include anything with a faceframe component. I live in the USA and 90% of cabinets have a faceframe and most all of our hardware and hinges are designed for compatibility with a faceframe. With that:

* Are there plans to add furniture with basic faceframe parameters?
* If not, do you accept pull requests to add new furniture?
* After looking at the code for magicStart, would you be interested in me doing a little cleanup to make future furniture inserts easier?

Thank you again for all of the work that you have put into this project. It is more complete than I ever could have imagined, and I plan on using this for the rest of my woodworking life.

dprojects, 2025-01-27:
> I absolutely love this package and am super excited to be using it for my wardrobe and custom cabinet projects. 
> Thank you again for all of the work that you have put into this project. It is more complete than I ever could have imagined, and I plan on using this for the rest of my woodworking life.

First of all, thank you for your kind words, my project is intended for exactly such people.

If you plan to create wardrobe, I would recommend you to see such movies: 
[How to design wardrobe quickly](https://www.youtube.com/watch?v=b2YR5ytC6fU)
[How to create wardrobe in real life](https://www.youtube.com/watch?v=PfuHzfqR4-M)
[How to make modular wardrobe - my recommended approach](https://www.youtube.com/watch?v=ASpLd4EVQ3g)

>  I have noticed that the magicStart furniture list doesn't include anything with a faceframe component. I live in the USA and 90% of cabinets have a faceframe and most all of our hardware and hinges are designed for compatibility with a faceframe. 

What kind of face frame you mean? some kind of front frame with glass? or just some kind of decoration at the front?
There is [front example with detailed clamp and decoration in example folder](https://github.com/dprojects/Woodworking/tree/master/Examples/Parametric#fronts) but is not linked at magicStart.

>     * Are there plans to add furniture with basic faceframe parameters?

Currently I am during making new tool magicGlue for adding expressions, I have it working for Cubes but maybe I will make it working for other objects as well, and create better glue bottle icon (a little spoiler :-D). 

And also I have a slight stomach problem, I don't know if I ate something or caught a cold, maybe next covid version, so I don't know when this tool will be ready. I can think about something new later, when I'm still alive :-D

>     * If not, do you accept pull requests to add new furniture?
>     * After looking at the code for magicStart, would you be interested in me doing a little cleanup to make future furniture inserts easier?

I want to keep the code clean, so I am not into accepting pull requests but you can add pull request so I will see what kind of changes you exactly mean and than maybe I will be able to make some improvements or add new items. 

Thanks, now I am going to create new magicGlue icon, I guess, and later I will look closer at the Pad constrains edit :-D 
Also I have to make little docs for it but maybe I will submit this feature in 2 steps, I don't know, first for Cubes and later for other objects...

awschult002, 2025-01-27:
> What kind of face frame you mean? 

Of course! Here is a [wkik](https://en.wikipedia.org/wiki/Face_frame#Construction) article describing a face frame in general, and here is a really good reference on building a very standard cabinet here in the USA.

https://www.woodshopdiaries.com/diy-base-cabinets-for-laundry-room/

That face frame servers many purposes. It allows a cabinet maker to sub-divide a large cabinet carcass easily. Drawers are often attached directly to the face frame and the back of the cabinet, which helps to divide up labor and reduce wood cost. Doors hinges in the US are also designed to attach directly to a 3/4 inch thick face frame and have specific reveal dimensions for it. 

The cabinets that you have available are effectively complete and nearly identical to a US one, except for the addition of a frame on the front. I would be happy to post more pictures of what kind of cabinets and hardware I have in my kitchen setup if you are interested.

> I am not into accepting pull requests but you can add pull request so I will see what kind of changes you exactly mean

I understand. I would be happy to post a PR with some changes. It might take me some time to get around to it; but I will try to get something soon.

dprojects, 2025-01-27:
> [wkik](https://en.wikipedia.org/wiki/Face_frame#Construction) article describing a face frame in general, and here is a really good reference on building a very standard cabinet here in the USA.
> https://www.woodshopdiaries.com/diy-base-cabinets-for-laundry-room/

OK, now I understand what you mean. Personally, I don't use it, I had something like that in the past in the kitchen, but I made new kitchen a year ago and now I don't have it. 

But I think I can add it and it could be done in the similar way as foot at megicStart, but in Z direction (vertical not horizontal like foot). And probably should be option to select start vertex point, for example left bottom corner, so the face frame would be created like front outside... but if you need exact pattern of such face frame, let me know...

> I understand. I would be happy to post a PR with some changes. It might take me some time to get around to it; but I will try to get something soon.

You don't have to post full working code, you can even describe it here, just for me to understand what you exactly mean... 

btw: I have magicGlue working with Pads as well and new icon :-) so I am going to make expressions clean option for Pads, some docs, and testing... make movie... and then I will submit it and then I will look into the face frame creation... so meanwhile you can post something here...

awschult002, 2025-01-28:
>But I think I can add it and it could be done in the similar way as foot at megicStart, but in Z direction (vertical not horizontal like foot).

by doing it this way, could you still acquire the cabinet dimensions from the selected object? or would the user have to input those dimensions in again?

I would think that you would want to make the face frame as part of the original cabinet build, that way you can guarantee dimensions more easily. Could it be a boolean option on the cabinet? include faceframe, or not?

awschult002, 2025-01-28:
i posted a PR. it's stripped down, but it at least ran on my machine. I would be happy to discuss more about what I had done if you are interested.

dprojects, 2025-01-28:
> by doing it this way, could you still acquire the cabinet dimensions from the selected object? or would the user have to input those dimensions in again?

This is the questions, because there are 2 types of face frame:
- inside - I had it in the old kitchen, it is better to fasten with screws and is more stable but may look bad with visible a gap, a furrow, between the main structure and faceframe, but here the depth dimension will be the same after faceframe added
- outside - I guess it is more common, it looks more beautiful from outside, you can add front inside and everything looks clear, but to fasten this you have to use glue probably to not show the screws... also the overall depth of the furniture should be smaller of the faceframe thickness... in this case user can create smaller furniture (for example depth - 18 mm) but need to know before...

So which one you need, outside or inside?
 
> I would think that you would want to make the face frame as part of the original cabinet build, that way you can guarantee dimensions more easily. Could it be a boolean option on the cabinet? include faceframe, or not?

I don't say no to cabinets with faceframe, probably I will make one as well, the most popular you want, but I don't want to create so many predefined furniture versions with faceframe...  

In my opinion faceframe should be some kind off additional thing, for example 3 versions:
- simple frame around
- simple frame around with center bar
- simple frame around with 2 vertical bars and 2 horizontal bars for drawers

So you will be able to remove or copy some bars at the last one and create more complicated patterns quickly... 

Also there is question for faceframe construction:
- do you use the top and bottom bars longer? or they should be inside the sides?

Also related to faceframe properties:
- I guess I will copy the selection method from front creation... 4 edges around to determine correctly the gap size?
- also there will be edit to wood thickness and width of the bars...
- I am not sure for other properties...?

> i posted a PR. it's stripped down, but it at least ran on my machine. I would be happy to discuss more about what I had done if you are interested.

Personally, adding new functions is not a problem for me. I simply copy a fragment of code from existing functions and modify the function that calculates and creates furniture, so it's quite fast. It may not be super ideal with all the possibilities of QT, but for now I somehow manage it.

I see that you know QT and programming in general. It's good to meet someone who can program. I personally don't have that much experience using layout and widget, but I downloaded this file and will analyze it to see if I can adapt something from it.

Also thanks for the idea about faceframe. I think this is good idea for new feature because the faceframe can be used to create front with glass. Personally I don't use glass in real life because I don't have tools to cut glass but many people love it and want it. Also this add to the furniture more elegant decoration look. So I think such feature will be very good.

awschult002, 2025-01-28:
You make a lot of very good points. And the more you explain about the face frames, the more I agree that face frame should be a separate tool that can be applied to existing geometry.

>So which one you need, outside or inside?
outside is all I ever really see (and my direct need)

>do you use the top and bottom bars longer? or the top and bottom should be between the sides?
also a great question. I don't think that it matters a whole lot. I am sure every manufacturer does it differently depending on the needs of the frame. I have a habit of putting runners on top of stiles to help with compression forces from above. but I could see an argument for keeping runners inside of stiles because if you want to divide the frame into stacked drawers, then you can use the same length of runner for all.

For me. I would just do runners on top and stiles inside. 

>I guess I will copy the selection method from front creation... 4 edges around to determine correctly the gap size?
that sounds like a great idea

> also there will be edit field for wood thickness and width of the bars...
ideally, there should be an edit for this. However, in the US, the hardware is designed for 3/4 default thickness

>I am not sure for other properties...?
The only other properties i can think of would be overhand/lip for each side of the cabinet. The frame is wider than the thickness of the cabinet wall, so how wide do you make the frame, and where does the frame sit on the side panel? I have built cabinets with the the frame flush with the outside of the carcass. Which makes putting cabinets next to each other a breeze. but this also creates a substantial lip inside of the cabinet that could result in plates or bowls hitting as you remove them. The cabinets in my kitchen have a small 1/4" lip on the inside, but overhang the outside of the cabinets by 12mm. This resulted in an annoying effort in lining up the cabinets and attaching side by side cabinets, as I had to be very aware that they were parallel all of the way down.

Adjusting the width of the frame can help to change the properties of the lip; but consider another property of the face frame is to help set the "reveal" or how much of the frame shows around the door. Some are full frame with no reveal and some are not. The hinges that you select are also designed for such a thing..... which makes me wonder is the width of a frame has been standardized in the industry..... I just wen all around my house and measured width from different manufacturers. they are all 1.5" wide (38mm). So i think we should start there. 

>  It's good to meet someone who can program. I personally don't have that much experience using layout and widget,
 I am not much of a python programmer. But I do program in other languages professionally. If there are any questions that you have about programming, please feel free to ask. I really enjoy talking about both programming and woodworking.

dprojects, 2025-01-28:
> You make a lot of very good points. And the more you explain about the face frames, the more I agree that face frame should be a separate tool that can be applied to existing geometry.

Yes. I think the faceframe should be an option you can add to the existing structure, such approach gives more flexibility and posibilities for design. However, I guess there should be also something predefined, ready for use, for those lazy ones like me for example :-)

> outside is all I ever really see (and my direct need)

OK so will be but how it is attached to the furniture? not glued from front?

> For me. I would just do runners on top and stiles inside.

OK.

> ideally, there should be an edit for this. However, in the US, the hardware is designed for 3/4 default thickness
>
>
> The only other properties i can think of would be overhand/lip for each side of [...]

Do you mean that it is a frame that is put on the sides from the front and there are additional thickenings on the sides, so you can't then perfectly push two cabinets next to each other? if so, that doesn't change the matter because there should also be an option to enter the thickness that sticks out to the sides, i.e. the thickness of the material from which such an overlay is made...

Do you have any photos of how it looks in practice?
Are there any photos of this on the internet...?

awschult002, 2025-01-29:
> OK so will be but how it is attached to the furniture? not glued from front?

Usually the frame is glued to the front of the carcass, sometimes with pocket holes. This can be seen at step 5 of the following link
https://www.woodshopdiaries.com/diy-base-cabinets-for-laundry-room/

>Do you mean that it is a frame that is put on the sides from the front and there are additional thickenings on the sides, so you can't then perfectly push two cabinets next to each other? 

No. I didnt do a great job of explaining it. I will post some pictures of the cabinets in my kitchen.

below, you can see the "lip" or overhang of the frame. This is because the frame is 1.5" wide and the wall of the cabinet is only 1/2" thick.
[image]

Here is another image of the lip on the inside of the cabinet

[image]

This lip causes the cabinet wall of an adjacent unit to not sit flush against. Note the 1/2" gap in-between

[image]

Although the gap isn't a huge deal. I don't think its entirely necessary, and I think I would prefer the frame to be flush with the outside edge.

dprojects, 2025-01-29:
> No. I didnt do a great job of explaining it. I will post some pictures of the cabinets in my kitchen.

I also always have such a problem to describe a given functionality with words as I create documentation. It is always better to see a photo, especially every manufacturer has its own standards.

Such a frame in the front can make more difficult the removal of dishes. The best use of such a frame is probably for such a cabinet, for example, under a sink, which has a lot of cutouts at the back, which makes the structure weaker. But if you add such faceframe, then the furniture will get stiffer and more stable.

Also in case of large wardrobes with two doors, something like that was used to place magnets on this middle bar, thanks to which the door did not fall inside and there was no gap.

But regarding the feature, I would see such options:

 Set by user:
- **Single bar width:** - this will be 38 mm by default
- **Single bar thickness** - this will be 12 mm by default (it is 1/3 of width) but I think it can be set by the user?

Calculated after **calculate faceframe** button press:
- **Start XYZ** - this will be starting point as usual...
- **Lip outside** - I don't know how to name it, and this can be calculated from single bar width, for example 1/3 of width = 4 mm
- **Depth sink** - this will be the value how deep the sides (furniture structure) sink inside the faceframe, router depth, for example in this case, 1/2 of thickness = 6 mm

This could be sample setup screen:

[image]

Regarding the furniture sample with predefined faceframe, it could be for example 900 mm (width) x 400 mm (depth) with single center bar and maybe single horizontal bar to be able to achieve any pattern by copy or remove... ?

What do you think? The defaults and calculations are ok?

awschult002, 2025-01-29:
* width = 1.5" (38mm) 
* thickness should be 3/4" (19mm). 
* outside lip = `0` for my needs, but nice to have the option to adjust
* inset (groove) = `0` for my needs

I think everything you got is really good. the only mention that i would make is. If you intend on providing a face frame with a center stile that divides the cabinet into 2 spaces; and you plan on putting doors over those spaces; then the center stile needs to be 2x width because the doors and hinges for doors are designed to stretch and cover the 38mm width of the frame. If you don't make it 2x, then the doors would interfere. 

Think of it like each space is its own box and its own full face frame. so when they meet in the middle the left and right sides touch and make a 2x wide vertical stile.

That is the only comment that I have left. If you don't want that, then I am not bothered either. 

You know... I have 2 builds that I am modeling right now and they both needs face frames. Perhaps I can create a youtube video of what I am doing and we can further discuss how I am screwing up the flow. and perhaps what can be improved.

dprojects, 2025-01-29:
>  width = 1.5" (38mm)
> thickness should be 3/4" (19mm).
> outside lip = `0` for my needs, but nice to have the option to adjust
> inset (groove) = `0` for my needs
> and make a 2x wide vertical stile.

I don't really believe that all the doors and drawers are supposed to be held together only by glue... I thought there was some kind of groove, that this frame was applied to the walls of the wardrobe... that's strange because there must be quite a lot of tension there on such glue... don't there happen some accidents where everything collapses? or starts to break after a few years when the glue starts to weaken?

But in such a case I think we should give the possibility to choose these parameters and then only the entire width and height will be calculated, and the starting point will already be calculated taking into account the width of the furniture board... then it will be very easy to draw these posts...

Something like this I think should be ok:

[image]

awschult002, 2025-01-29:
> Something like this I think should be fine?

I think it looks great!

> The top and bottom have little offsets, in this case 18 mm (wood thickness), so I think this can be some kind of base for predefined sample with face frame, this is the most used in US? or something other?

The expensive, manufactured cabinets that have gone into my kitchen (see images above) DO extend the sides the FULL height of the cabinet + face frame. That would mean that the sides are flush with the edge of the face frame and do not extend passed. This can be seen in my photo here:
![cabinet](https://private-user-images.githubusercontent.com/73046001/407568828-a78406a2-bc81-4dc4-a95b-fda4cb308487.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzgxODQ5OTYsIm5iZiI6MTczODE4NDY5NiwicGF0aCI6Ii83MzA0NjAwMS80MDc1Njg4MjgtYTc4NDA2YTItYmM4MS00ZGM0LWE5NWItZmRhNGNiMzA4NDg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTI5VDIxMDQ1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZkM2U3ZDc0NjZhMWFjMzVkMTg0NjNhNWI3MTZkNWY0OWNkOTM5Mjc4MmQ5YTA3ZjU5NjBmNTM1NTM5ODcxZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fh9G17JDXQFLKuizMf-j8UzJ1MIHgOeg-rG6wceIMlw)

That extension is not a requirement that I was going to put on you, but if you want to add it in; then I certainly will not complain.

dprojects, 2025-01-30:
> I think it looks great!

I just have [submitted the Face Frame feature](https://github.com/dprojects/Woodworking/commit/30b39ca1d91114bd21e39f0a951c6cc0f373a3a6). I also changed the functions places, this is why there are so much changes... 

Here is the movie about this feature: [How to add Face Frame to the furniture](https://www.youtube.com/watch?v=CtWfvxd4UmI)

Let me know what do you think. Something should be added or changed?

> The expensive, manufactured cabinets that have gone into my kitchen (see images above) DO extend the sides the FULL height of the cabinet + face frame. That would mean that the sides are flush with the edge of the face frame and do not extend passed. 
> You can also see that the back panel of the cabinet extends to meet the lip of the face frame as well. This gives the installer a place for mounting screws without going inside of the cabinet. It also allows the cabinet to sit flush on the bottom or top and not have to worry about the face frame getting damaged as they are flush together.

I have to watch YT movies about the US solutions because it sounds very strange to me :-)
However, thanks for the feature issue because this will be good base for next feature - front with glass :-)

awschult002, 2025-01-30:
I just watched the youtube video and I think what you have done is great! the extra options for center and horizontal dividers is fantastic. It makes extremely quick work of designing cabinets and I am thrilled that you were able to put this together.

There is just one _minor_ issue before we close this out. The sides of the cabinet are **not** supposed to extend passed the frame. 

[image]
The part that your mouse is hovering over needs to be flush with the frame.

Making the sides flush with the frame would prevent this big gap from occurring when you stack 2 cabinets. 

[image]

This is the last change that I would request. Then I believe this is complete. Thank you so much for your work.

dprojects, 2025-01-30:
> The sides of the cabinet are **not** supposed to extend passed the frame.

I think I saw something like that on YT and that's why I was inspired by it :-)

However, I would leave it like that because the offset is the thickness of the board, i.e. 18 mm, a few clicks are enough for it to disappear. In my opinion, it would be better to add a new type of sample cabinet based on a different construction, let's say with an top and bottom bars through the whole width, sides inside, or some other type of construction that you think is the most popular in US, then there would be greater possibilities for potential adaptation to your own needs. Everyone could choose the initial piece of furniture that will give them the fastest possibilities for later reconstruction.

For now, I think I'll focus on a slight improvement of magicMeasure because it annoys me a bit that sometimes I have a problem clicking the vertex, and in the meantime let me know what furniture construction would be the most valuable in your opinion? and then I will add it in some time, maybe tomorrow or later... 

It is best to post a photo of such furniture, or a link from the Internet, I guess...

awschult002, 2025-01-30:
Fantastic! I will play around with this new feature for my designs and I will also do a little more research on how other cabinets in the US are manufactured, as well as how DIY youtubers are making their designs.

dprojects, 2025-01-30:
> Fantastic! I will play around with this new feature for my designs and I will also do a little more research on how other cabinets in the US are manufactured, as well as how DIY youtubers are making their designs.

OK. You can create new issue if you find something interesting worth to add.
But regarding the face frame feature you were right, if you change the sides sizes the overall furniture size will not be as desired at setup screen, so I fixed it... there are also bug with furniture overall depth and starting position, now everything should be fine:

[image]

I also decided to add the storage construction with 2 horizontal bars: 

[image]

closed 2025-02-02 by dprojects