dprojects/Woodworking #54: Woodworking on FreeCAD-LinkStage3 version.

MoonLoi, 2025-02-05:
Hi! I'm using [FreeCAD-LinkStage3](https://github.com/realthunder/FreeCAD), but this version of this workbench doesn't fully work. MagicMove and MagicGlue end with the error: 

`'Gui.View3DInventor' object has no attribute 'getCornerCrossSize'`

Is it possible to fix this version of FreeCAD? I really like this workbench. And I really like FreeCAD-LinkStage3.
I use Archlinux. The native version of app image 0.21.2 does not launch. GLX cannot be initialized.

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: a9810d509a6f112b5ac03d4d4831b67e6bffd5b7
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * BillOfMaterials 1.0.2
  * FreeCAD-themes 2025.1.7
  * freecad.optics_design_workbench 0.1.5
  * Behave-Dark-Colors 0.1.1
  * Defeaturing 1.2.3
  * PieMenu 1.6.0
  * OpticsWorkbench 1.1.0
  * Woodworking 2025-02-04 0.21.2.33771.0.23.1
  * kicadStepUpMod 11.5.2
  * ThreadProfile 1.92.0
  * Silk 0.1.5
  * CurvedShapes 1.0.13
  * 3D_Printing_Tools
  * sheetmetal 0.7.5
  * MeshRemodel 1.10.36
  * woodworking 0.21.2.33771.0.23.1 (Disabled)
  * fasteners 0.5.37
  * Freecad-Built-in-themes-beta 1.2.2
  * Curves 0.6.55
  * OpenTheme 2024.9.1
```

dprojects, 2025-02-05:
> Hi! I'm using [FreeCAD-LinkStage3](https://github.com/realthunder/FreeCAD), but this version of this workbench doesn't fully work. MagicMove and MagicGlue end with the error:
> `'Gui.View3DInventor' object has no attribute 'getCornerCrossSize'`
> Is it possible to fix this version of FreeCAD? 

Probably there is different API and there is no `getCornerCrossSize` function. It refers to right bottom cross at the 3D screen. So if there is no such thing there might be more problems during using this workbench in such configuration.

> I really like this workbench. And I really like FreeCAD-LinkStage3. 

Thanks, As far as I know the Realthunder has better lightning and shadow options but I didn't use it for real.
Maybe if I will find some time someday I will look closer to this branch and will try to see how much differences in API there are, probably a lot...

> I use Archlinux. 
> The native version of app image 0.21.2 does not launch. GLX cannot be initialized.

It should... there is also FreeCAD 0.21.2 version for Archlinux, I guess here: [FreeCAD-0.21.2-Linux-aarch64.AppImage](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-aarch64.AppImage)

I'll leave this issue open but I don't promise it will be resolved. However, Realthunder's version is very interesting, he must be smart one :-)

MoonLoi, 2025-02-05:
Thanks for the answer!

> It should... there is also FreeCAD 0.21.2 version for Archlinux, I guess here: [FreeCAD-0.21.2-Linux-aarch64.AppImage](https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Linux-aarch64.AppImage)

No. This AppImage is ARM64 architecture.

dprojects, 2025-02-05:
> No. This AppImage is ARM64 architecture.

Maybe you should try Xubuntu? I use it... 
However, if you use such customized linux version, you should probably be able to fix it on your own, just comment lines of code that refers to the cross screen settings, so you should be able to run it without cross switchers, but for sure there are more API differences, so there is need to test all features to tell where are the differences... to be honest, there is no time for that, maybe someday...

dprojects, 2025-02-25:
Hello again... I was looking at this issue and looks like there is no `FreeCADGui.ActiveDocument.ActiveView.getCornerCrossSize()` reference... also I was looking for the `getCornerCrossSize()` elsewhere but I wasn't able to find it... so I decided [to turn it off if there is no such reference](https://github.com/dprojects/Woodworking/commit/4b7b7f0c61a0faa26ca3501a3c878c05e1f4b74d) also I created [issue at Realthunder repository](https://github.com/realthunder/FreeCAD/issues/1060) so you can follow this issue there... if there will be such thing I will add support for it...

I will close this issue but if you find other issue during using the LinkStage version open new issue or reopen this one... 
Officially can't support this branch right now because it requires all feature testing and there is a lot of them...

closed 2025-02-25 by dprojects