dprojects/Woodworking #56: Support for Assembly

dprojects, 2025-04-21:
> I don't know if this falls into the bucket of 1.0 support or not; but for my workflow, I really really need assemblies. Is there anyway to get the containers for pre-made furniture into a freeCad Assembly structure? This would allow super fast work of TechDraw blueprints and exploded views.  

 _Originally posted by @awschult002 in [#49](https://github.com/dprojects/Woodworking/issues/49#issuecomment-2818448639)_

dprojects, 2025-04-21:
I personally don't use Assembly, because for me it doesn't make sense to draw every board in Sketch and then create an assembly. This may be because I have no experience with CAD programs, FreeCAD is my first program, I launched Blender once before but I had no idea how to quickly create a piece of furniture and the list of cuts I needed.

In my opinion, when it comes to furniture design, a design method based on unrelated elements works much better, which gives you huge possibilities for rebuilding any piece of furniture.

As for parameterization, this is not a modeling method in my opinion. I used to struggle for months with a spreadsheet and creating mathematical expressions, and then it turned out that all the parameterization I did didn't allow me to make the changes I wanted.

I look at FreeCAD more from the perspective of programming, and that's what's really cool about it, that you can program whatever you want. Parameterization is a cool addition in my opinion, but the truth is that FreeCAD is as parametric as the mathematical formulas you come up with yourself, there is no automatic parameterization here, I personally thought about introducing it, but for now it is on-demand parameterization using magicGlue.

The situation is different when it comes to designing engines or artistic furniture, where each board or element is different and has to be made separately. In such cases, such a workflow based on Assembly is absolutely necessary.

Instead of moving elements apart to see a part, it uses hiding with the "space" or making all elements transparent. This is the advantage of unrelated elements, that you can remove, modify, or temporarily hide each element to have easier access to other elements. In my opinion, this approach works perfectly when designing cabinets.

As for TechDraw, I only use it to print a cutting list. In general, I do not understand why the documentation is not generated automatically based on the model. If all dimensions can be done on the model, why should you draw everything from scratch additionally on TechDraw pages?

This turned out to be a long article and I don't know if my approach will be met with enthusiasm :-)

Personally, I still plan to add some better support for Assembly, but first of all, I don't know anything about Assembly, I have one example and I added support for Assembly in the cut list based on this example, but I don't know how someone created this Assembly... I was supposed to read the description some time ago but somehow I still don't have time to deal with it... actually, I don't really know what people expect when they say there should be support for Assembly?

Maybe if more people here start writing what they expect and give some simple examples, then I will know... and I will add some better support.

dprojects, 2025-04-22:
> > I personally don't use Assembly, because for me it doesn't make sense to draw every board in Sketch and then create an assembly. This may be because I have no experience with CAD programs, FreeCAD is my first program,
> The way traditional CAD packages work, is they expect "parts" and "assemblies", where an assembly is the 3D placement and constraint of many "parts". So in the context of wood working; each board is a "part" and each furniture would be an "assembly". This is nearly identical to what you have already done with your containers. Assemblies are just containers that FreeCAD provides much more capability for ( in version 1.0 ).

I think in the same way but... some people use only Body and Pad without Part and LinkGroup containers. I agree also this might be a problem for people from CAD world, they looking for Assembly and are not able to accept different workflow. 

> I am making a huge project made up of 11 different cabinets. I want blueprint drawings of each one. 

Last year I made my kitchen, 5 kitchen wall cabinets (5 doors) + 5 normal cabinets (4 drawers, 5 doors, all with soft close) - I made all of them one by one, using the cut-list only... maybe I will show someday the full construction... I have no idea what the blueprint is, you have any example? why TechDraw not create it automatically if this is standard? 

> every blueprint, instruction manual, and Ikea pamphlet has and exploded view. This allows someone to print 1 page and see how all of the parts fit together. they are critical.

But it is hard to create, especially for beginners, I am not sure if anyone can do that just like this... In my opinion documentation and parameterization should not be next long design process, this should be automatic with single click...

> Here is an explicit example of what I need to do https://youtu.be/k51YFdXIIek?si=dpw_gmB_7iZJlbdM

I like the exploded view but I don't like it is done manually... it should be automatic with single button click... 

> So I would have to rebuild every single cabinet out of the container, into an assembly with the proper origin for TechDraw to render it correctly. 

I guess, you can currently move out pieces from containers:

**Step 1:** convert all panels from Cubes into Pad objects via [panel2pad](https://github.com/dprojects/Woodworking/tree/master/Docs#panel2pad)
**Step 2:** move out all Body objects from Part container via [selected2Outside](https://github.com/dprojects/Woodworking/tree/master/Docs#selected2outside)
**Step 3:** create assembly object and this part I am not sure because I don't know the Assembly, I have already installed Assembly4 but I don't know how to use it... please see the attached file sample, this is FreeCAD 1.0 file, and create Assembly there and write here what buttons did you clicked in Assembly4 workbench to create Assembly between those two pieces of wood... (rename the file without zip)

[create_assembly.FCStd.zip](https://github.com/user-attachments/files/19852896/create_assembly.FCStd.zip)

awschult002, 2025-06-12:
I saw your video! Thank you so much!