dprojects/Woodworking #57: Error when copying with MagicMove using US units of measure

ctefer, 2025-04-27:
When using MagicMove to copy along an edge I get the following error.

Traceback (most recent call last):
  File "/home/tefer/.local/share/FreeCAD/Mod/Woodworking/./Tools/MagicPanels/magicMove.py", line 1648, in createCopyByEdgeX
    self.createCopyByEdge("X")
  File "/home/tefer/.local/share/FreeCAD/Mod/Woodworking/./Tools/MagicPanels/magicMove.py", line 1025, in createCopyByEdge
    offset = MagicPanels.unit2value(self.cbe5E.text())
  File "/home/tefer/.local/share/FreeCAD/Mod/Woodworking/./Tools/MagicPanels/MagicPanels.py", line 4412, in unit2value
    forCalculation = Units.Quantity(str(unitString)).getValueAs("mm")
ValueError: Unit mismatch

Error appears to be due to America.

FreeCAD Version: 0.21.2
Woodworking Workbench: 0.23

To recreate. Change preferences of FreeCAD to Building US unit system. Edit->Preferences->General->Unit System-> Building US (ft-in, sqft, cft)
As shown in Video "Plant Box with Trellis":

1) create a bottom panelXY, move panel down
2) create a side panel panelZY
3) open magicMove
4) refresh selection for panelZY
5) copy to edge
6) set panelXY, Edge 10
7) "Copy along X"

An issue will be generated in the report view.

ctefer, 2025-04-27:
Workaround is to use the Unit System "US Customary".

Only supported units of measure are "Standard (mm)" and "US Customary". My guess is that this would crash if any other preferences are used. Perhaps a future feature.

ctefer, 2025-04-27:
Sorry, bad at git

dprojects, 2025-04-28:
Hi, I got several e-mails from github about open issue, closed, reopened, closed :-) and it was very funny to read :-)

> Change preferences of FreeCAD to Building US unit system. Edit->Preferences->General->Unit System-> Building US (ft-in, sqft, cft) As shown in Video "Plant Box with Trellis":

I am not expert from Units and I am not sure which one should be set in US, but I use "US Customary" because it accept short notation like " (quotation mark) and others accept only ft (feet). So this might be FreeCAD bug with Units if the correct units US setting should be different... 

Currently I am going to eat dinner and later I am going to finish next feature (explode view and TechDraw docs maybe) and if I finish the feature I will look into the units issue...  but meanwhile let me know what kind of units you expect in US? and what kind of notation should be accepted?

dprojects, 2025-04-30:
> Error appears to be due to America.
> FreeCAD Version: 0.21.2 Woodworking Workbench: 0.23

Thanks for this issue because it is FreeCAD bug available also in FreeCAD 1.0 stable.

For "Building US" units schema the FreeCAD converts "0 mm" into "0" not into "0 in" like for all others schemas:

```
 >>> from FreeCAD import Units
 >>> Units.schemaTranslate(Units.Quantity("0.0 mm"), 5)
  ('0', 25.4, 'in')
 >>> Units.schemaTranslate(Units.Quantity("0.0 mm"), 2)
  ('0,00 in', 25.4, 'in')
 >>> 
```   

this FreeCAD bug is even more complicated, better trap I would say, because it appears only for "0 mm" value at this schema:

```
>>> Units.schemaTranslate(Units.Quantity("0.0 mm"), 5)
('0', 25.4, 'in')
>>> Units.schemaTranslate(Units.Quantity("500.0 mm"), 5)
('1\' 7" + 5/8"', 25.4, 'in')
>>> 
```

> Workaround is to use the Unit System "US Customary".

You could also change manually the "0" value in "Additional offset:" into "0 in" but [I added fix to the library](https://github.com/dprojects/Woodworking/commit/fa740e49db00091c8ae67b686d8b946eae62f6c5), so now it is working... 

> Only supported units of measure are "Standard (mm)" and "US Customary".

and you were right, the "0 mm" values worked only with those schemas due to the FreeCAD "Building US" trap...

closed 2025-04-30 by dprojects