dprojects/Woodworking #62: addon manager git clone fails on 1.1 snap

leho-l, 2025-07-16:
Git returned a non-zero exit status: -1
Called with: /snap/freecad/1574/usr/bin/git clone --recurse-submodules https://github.com/dprojects/Woodworking /home/USER/snap/freecad/common/Mod/woodworking

Returned stderr:
Cloning into '/home/USER/snap/freecad/common/Mod/woodworking'...
fatal: early EOF

dprojects, 2025-07-16:
> When installing from addon manager the following error ocurres: Git returned a non-zero exit status: -1 Called with: /snap/freecad/1574/usr/bin/git clone --recurse-submodules https://github.com/dprojects/Woodworking /home/USER/snap/freecad/common/Mod/woodworking
> 
> Returned stderr: Cloning into '/home/USER/snap/freecad/common/Mod/woodworking'... fatal: early EOF
> 
> when looking into the Mod/woodworking/ folder it contains .git folder with the files. THe foiles should be in the woodworking folder, not in a sub folder?

This issue has been opened before, but [it has been closed](https://github.com/dprojects/Woodworking/issues/60) because it is not a problem on the Woodworking workbench side.

The problem is multi-level:
- The basic problem is a **incorrectly written Add-on Manager** that downloads a `.git` folder instead of a [zip repository](https://github.com/dprojects/Woodworking/blob/8b6de031e374e6c2e8f4d33d12952a519b2f244f/Tools/debugInfo.py#L5) like I do in my update tool.
- An additional problem is related to `git` tool itself. Because in case of binary files the `git` tool, drastically increases the size of the `.git` folder, which should never happen. This is simply an incorrect approach.
- However, [before deleting the video files](https://github.com/dprojects/Woodworking/commit/c1db46b7efa17ea8898dd51b03280559279f7c9c), the overall Woodworking workbench size was over 200MB, and I tested the Add-on Manager in the latest FreeCAD 1.1dev and didn't experience any installation issues. So, it's likely a user-specific network configuration issue, as well.

I can leave this issue open, but probably **nothing more will be done** about it. I don't know if I want to filter `.git` history to meet any size limits, because over time the size might grow anyway and **the problem will reappear.**

leho-l, 2025-07-16:
Ok, thank you for your swift reply.
I manuaaly copied the addon, and  it seem to work.
You have done a huge ammount of work with this workbench.
Do you knowhow big is the userbase now?

Regards,

Leho Laaser
Pillitöökoda Härmoonik OÜ
tel.5541 079

easyw, 2025-07-22:
@dprojects 
unfortunately I have to agree so much about your review of the open source drift.
Regards
Maurice

dprojects, 2025-07-22:
> Unfortunately, I very much agree with your analysis of the open source drift.

Yes, this is sad. Indeed, someone unfamiliar with the FreeCAD community might wonder why someone like you, who has so many popular FreeCAD add-ons, not have his own topic section on the FreeCAD forum, not have administrative rights there, is not promoted by the main founders, is not considered as an expert in this field and not decide about developing this field.

However, this doesn't surprise me personally. But, I created my add-ons for myself, not for these people, and in a sense, I still feel guilty about attracting unsuspecting people to such a community. This is why it would be the best option to create my own product and community, one based on respect for independent talented developers.

Because, as you can see, even GitHub TOS, which is legally illegal, doesn't mention copyright or any recognition of the main creators. I understand that Microsoft wants [to protect the interests of corporations and everyone who relies on open source](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo#use-someone-elses-project-as-a-starting-point-for-your-own-idea), but unfortunately, it does so at the expense of creative people and, in a sense, [brings legal problems to forks that will remain despite the license change](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/what-happens-to-forks-when-a-repository-is-deleted-or-changes-visibility#deleting-a-public-repository).

So, as you can see, the problem is much broader, mostly political. But don't worry, probably soon someone will copy [Linus solution](https://forum.freecad.org/viewtopic.php?p=837836) into main FreeCAD branch [for grants](https://fpa.freecad.org/programs/fpadf-announcement), and then the community will be fixing for free what others broke for money.

dprojects, 2025-08-13:
> I got this on my desktop PC while on my work laptop Woodworking WB installed fine, both have 1.1 dev.
> 
> Turns out clicking the "Refresh local cache" helped and the add-on installed fine after that. I know this issue is closed, but in case someone else bumps into this and googles for an answer, maybe this is helpful :-)

Caching in Add-on manager is not good solution generally, but if someone find it useful, it is fine, but never should never be set as default. You can see another issue which might be also related to Add-on Manager caching: [Can't find tool after installation](https://github.com/dprojects/Woodworking/issues/68)