dprojects/Woodworking #66: magic Move Suggestion for usability improvement.

accspert, 2025-08-06:
Hi Darek,
as promised, here’s my first suggestion:

When opening MagicMove, the window currently stays on top of other applications rather than staying within FreeCAD. It might be more user-friendly if it remained within the FreeCAD environment:

[image]

The sentence "select object to move or copy" is clear and helpful. One idea could be to automatically update the selection when an object is clicked, without needing to press the "Refresh Selection" button. That button might instead serve better as a "Clear Selection" function. This way, users see updates instantly as they select objects.

Let me know what you think—happy to discuss or adjust anything.

dprojects, 2025-08-06:
> the window currently stays on top of other applications rather than staying within FreeCAD. 

Personally I don't like FreeCAD combobox window so I don't want to dock it into this by default :-) but maybe there should be additional option to dock it there... however if the window stays on the top you can always copy data into such window, for example dimensions from notepad or something like that... so in some case this might be useful...  

> automatically update the selection when an object is clicked, without needing to press the "Refresh Selection" button

I am not sure about it because currently you can have opened 2 tools, for example, and you can for example move and rotate different objects at the same time and you don't want to load currently selected object to rotate for magicRotate and also into magicMove for move... if so, I would solve this problem using another button, some kind of observer, if you click it this will be turned on and listen for you clicks... but I am not sure if this function is so desired and "blocking" from user perspective...

accspert, 2025-08-06:
For the first point, it seems a bit unusual compared to other software, so
it might be best to leave it in FreeCAD. I was confused when I saw it the
first time. I always try to consider the experience of beginner users.

Regarding the second point, having multiple selection windows can be
confusing. Keeping the interface simple would likely offer a better user
experience.

accspert, 2025-08-06:
> I have to rethink those two things, so I leave this issue open, because I
> don't want to break current possibilities. First user experience mostly
> depends from other CAD program experience. Different will be for someone
> from AutoDesk Fusion, and different from Blender or from other software or
> even without earlier experience, like I did, because FreeCAD was my first
> CAD program.
> Meanwhile I found bug in Woodworking workbench 1.0 automatic translation
> drop-down menu tool... so I renamed the translation branch into 2025.06
> now will be working fine if you will add German translation there...
> Also there is new repository with installation package
> <https://github.com/dprojects/Woodworking-package>... the magicCAD_1.0
> AppImage. This is installation package with FreeCAD kernel and Woodworking
> workbench <https://github.com/dprojects/Woodworking>. This AppImage will
> copy the Woodworking workbench into ~/.local/share/FreeCAD/Mod folder, will
> run the FreeCAD, and set the Woodworking workbench as default. So you will
> be able to update later the Woodworking workbench via debugInfo

Now only came in my mind that there is no button needed at all what belongs
to selection. The user could simply click on an object and move it, that's
it, new selection new movement. There is only the existing text needed
select an object and move it. And then a text that shows what is selected
currently.

dprojects, 2025-08-06:
> there is no button needed at all what belongs to selection. 
> The user could simply click on an object and move it, that's it, new selection new movement. 

There are [arrows](https://github.com/dprojects/Woodworking/tree/master/Docs#arrows) for that. If you select any supported object and click arrow button the selected object will move with thickness step.

But I agree your first experience feelings regarding FreeCAD in general. FreeCAD is hard to use. I remember first time I wasn't able to create simple piece of wood and move it to the exact place. Also for many time I didn't know what is the "Transform" option. In my opinion "Transform" should be renamed to "Move & Rotate". If you want to move object and you click right mouse button and see "Transform" you think this is much more complicated operation... not only move and rotate.

> the window currently stays on top of other applications rather than staying within FreeCAD.

And also regarding this issue. I use this functionality always after I create new tool with GUI for making documentation. I can open the new GUI and Kwrite editor at the same time and describe each widget there. So I think this option is useful. But maybe docking window option is worth to add as well but I need to think how to do it correctly.

closed 2025-08-10 by dprojects

accspert, 2025-08-12:
> Hey Darek how are you doing?
>
> Hi, thanks I am fine, I improved the globals management during those
> features, and also improved the makeAPI generator
> <https://github.com/dprojects/Woodworking/blob/master/Docs/makeAPI.py> so
> now the API docs
> will be looking a little better, and also I released new installation
> package based on FreeCAD 0.21.2 kernel
> <https://github.com/dprojects/Woodworking-package/releases/tag/0.21>...
>
> After an update all tool windows are disappearing when you click a other
> object. Now you click a object then you have to get the window back and
> then you can go on that's more clicks than needed. Can't it be that the
> window stays always on top in freecad but not in other applications? So it
> stays always in free cad.
>
> It is working as it should, the window stay at the FreeCAD level... you
> can see it here in the new video:
> New magicSettings options <https://www.youtube.com/watch?v=Y7Z3tQRZPn8>
>
> if you have problem with this it might be problem with your system
> configuration... because if you change the option to "no" the window from
> tools have not special window settings, so the system settings are there...
>
> Like the Properties or Task Windows
>
> there are no plans currently to put all windows into FreeCAD combobox
> because there are too much changes, FreeCAD has different approach there,
> so I can't say if someday such option will be available...

Hey Darek,

yes now it is working like in the video.

I’d like to suggest an enhancement for *magicResizer* that would improve
both usability and workflow:

*Updated instructions (displayed at the top):*

   1.

   Select an edge.
   2.

   Hold the Ctrl key and select a face, edge, or vertex.

*Proposed change:*

   -

   Remove the refresh button. When the user selects an edge and then a
   face, display the selection directly in a text field without extra clicks.
   -

   While the refresh button was intended to handle hidden objects, this use
   case is extremely rare in practice. Eliminating it reduces interface
   clutter, shortens interaction steps, and makes the tool more intuitive for
   the majority of users.
   -

   The new approach aligns with common FreeCAD selection workflows, making
   it easier for new users and faster for experienced ones.

*Regarding the “always on top” window:*

   -

   Since documentation creation for the woodworking workbench is a niche
   task, it may be more user-friendly to keep this window embedded inside
   FreeCAD instead of forcing it to remain on top. This would avoid
   unnecessary interference with other  tasks.

accspert, 2025-08-13:
> The problem is that when a window is opened inside FreeCAD, FreeCAD
> controls what happens with that window, and this is unfortunately not
> good...
> if the window would be embedded, there would be much more problems with
> resizing, text overlap, or even not working clicking or unloading
> objects... so it is asking for problems...

How about the other stuff in the previous email?

dprojects, 2025-08-13:
> I saw some windows in other work benches where the window just stays
> in freecad. For example in draft workbench to set up variable set.

... the Draft workbench is part of FreeCAD main branch and part of the FreeCAD but Woodworking workbench is additional sets of tools and nobody is forced to use it... 

> Or task window in part design. 

but you find it handy? good visible? 

> There are several other examples.

yes, maybe but Woodworking workbench set its own rules and many people love it... 

> But I don't know any software in the world where the window is handled like
> a standalone application. 

it is not standalone application because it is not working without FreeCAD and it is called from toolbar via icon... the window needs to be movable... that's the point.