dprojects/Woodworking #67: Help make cutout in simple shelf

Beep6581, 2025-08-09:
Hello

Respect and appreciation for your Woodworking workbench, what a great and feature-full tool!

I am trying to create a simple shelf, using dado joints, where the shelves are linked so that if I resize one shelf the others resize as well. I read some of the Woodworking documentation and followed a few videos. This is my second Woodworking project, but the first one I want to do "properly" ;)

I used the `magicMove` tool with the "Link" copy method, so that the copied shelves follow the master shelf.

The shelf:
[image]

It should have this curved cutout going through both left and right panels:
[image]

I run into a few issues:
1. I believe I have to use the "single panel cut by many knives with copies" or "single panel cut by many knives with links" tool to make the dado cuts. I followed the "[Quick Dado joints](https://www.youtube.com/watch?v=vMwfU3sup2k)" video, but it's old and you cannot see whether that old button was "with copies" or "with links". Regarding [magicCutLinks](https://github.com/dprojects/Woodworking/tree/master/Docs#magiccut), the docs say, "This tool works with the same way as magicCut tool but creates LinkGroup container for cut panels, knives, and uses container links for cut operation. Thanks to this approach you can change Cube to Pad or even add new element to the LinkGroup container and the cut will be updated with new content." Is `magicCutLinks` the right tool to use?
2. How should I sketch the cutout? Regardless whether I create the sketch on the `Cut, Right 004` item or on the `Right` item, I always run into a `Cut, Right : Linked object is not a Part object` error:
[image]

FCStd file (remove .zip extension):
[electronics shelf 800x900.FCStd.zip](https://github.com/user-attachments/files/21702590/electronics.shelf.800x900.FCStd.zip)

Any help much appreciated.

dprojects, 2025-08-10:
...and also I created sample structure from your sample file. There is also cut-list, exploded custom view but the "objects" TechDraw export not works with LinkGroup at FreeCAD, so I would rather recommend to go into screenshot version for TechDraw export. This "workflow" is hard, for sure not good for beginners because for further processing such model you need to know what to do to avoid FreeCAD "unsupported" structures... 

but advantage of such parametric approach is that, if you add or move any shelf the cut will follow the new structure, so if you will copy the shelf the cut will be created automatically, also if you move the up or down... 

[electronics_shelf_structure.FCStd.zip](https://github.com/user-attachments/files/21705103/electronics_shelf_structure.FCStd.zip)

dprojects, 2025-08-13:
If there are no answer, I believe this is already explained, solved... you can reopen this issue if there is problem and you need help. Thanks.

closed 2025-08-13 by dprojects

Beep6581, 2025-08-14:
@dprojects thank you for the sample project. It will take me some time to go through it with understanding.

The sample project is missing the curved cutout. I achieved that using the Part workbench with a sketch on surface, then extrude, then a boolean cut. Is that the correct procedure or is there a cleaner way?

[image]

Beep6581, 2025-09-08:
Sorry to bother you, but could you show how to make that cut-out using PartDesign?

dprojects, 2025-09-10:
@Beep6581 

Sorry I haven't replied, but I made several models of your design, using different approaches, and while creating the model in PartDesign, I realized that creating countersink holes for the shelves was too complicated. So I decided to [slightly modify the magicDriller](https://github.com/dprojects/Woodworking/commit/08610f6d775a794fa19bed96fe32796cc16920a1) tool and add separate buttons for positioning and drilling. This allows you to create these types of holes much faster now: [Drilling countersinks for shelves](https://www.youtube.com/watch?v=rd2W-L6OHuo)

> The goal was to see how far I could get with low-quality construction wood and breaking all the rules. I joined two 120mm-wide planks side by side to get a cheap 240mm-wide plank, using dowels and glue. It actually worked! Nothing warped, since each piece holds in place every other piece. 

When it comes to joining boards together to obtain a larger board, this can be easily reproduced, in fact, this is how it should be designed to obtain the correct cutting list for all elements:

[image]

Regarding the approach, I created three models:

- **Part::Box** - these are simple panels made with overlapping joints, without cutting out the joints. If you're building your own furniture, I highly recommend this version. This approach isn't very detailed, but it's very quick and easy to implement, and the cutting list will be accurate. In this particular case, if you specify the distances between the shelves, it will be clear enough where the shelf grooves should be.

[Bookcase_03_Part_Box.FCStd.zip](https://github.com/user-attachments/files/22258402/Bookcase_03_Part_Box.FCStd.zip)

- **Part::Cut** - in this case, you can also cut the kick-toe using a simple block. However, with this approach, you won't be able to cut the countersink holes without using the PartDesign approach.

[Bookcase_04_Part_Cut_parametric_way.FCStd.zip](https://github.com/user-attachments/files/22258414/Bookcase_04_Part_Cut_parametric_way.FCStd.zip)

- **PartDesign** - this is the most detailed approach. With this approach, you can achieve all the details, including screws and pilot holes for screws.

[Bookcase_05_Part_Design.FCStd.zip](https://github.com/user-attachments/files/22258423/Bookcase_05_Part_Design.FCStd.zip)

> I wanted to plane those 240mm planks to get them perfectly flat and thinner, but only after joining them did I discover that my planer is not wide enough 😂 So it was all just sanded down to hide the seams. It's not something one would sell, but I'm perfectly happy with it for personal use.

It turned out beautifully. I personally like this light wood color, pine I guess? it reminds me of a kitchen table on a sunny day. My Mom, on the other hand, likes the color of mahogany furniture, but I actually hate mahogany, I prefer more light.

Real wood is very problematic, you need the right tools, place to play with dust, you need to know what you're doing, and the wood must be properly prepared. For this reason, furniture is most often made from cut-to-size chipboards.

In my opinion, real wood furniture is more decorative and luxurious than practical. When you're making joints, maybe it's worth highlighting them somehow, for example, painting them in different colors to make the joints more visible?

[image]

Additionally, maybe it's worth drilling counterbores holes for screws and driving in wooden dowels to cover the screw heads?

[image]

Beep6581, 2025-09-10:
Amazing, thank you! I will work through all the info, it will take some time. 

The most important benefit of using FreeCAD for woodworking for me is that I can get a tech drawing with dimensions for the dado cuts. Example from my own attempt:

[image]

It seems easy to make a drawing like that using your simple Part::Box technique, that's great. However, when I make an orthographic projection, it turns out a bit corrupt. I thought I'd ask whether this could be a Woodworking Workbench bug before I report it upstream (I am using the latest self-compiled main branch):

[image]

Beep6581, 2025-09-10:
> I'm not sure exactly what error you're referring to

See the red arrows in my screenshot.

I will look through the new files this weekend, thank you.