dprojects/Woodworking #70: Error Message Wire2Pad

accspert, 2025-08-13:
Hi Darek,

steps to verify:

- create a sketch and a wire.
- click wire2pad
- see error message in the buttom.
attached is the message.

[image]

accspert, 2025-08-13:
Looks like it is alaways hapenning when you create a sketch, no matter of wire2pad

dprojects, 2025-08-13:
To use [wires2pad](https://github.com/dprojects/Woodworking/tree/master/Docs#wires2pad) you need to keep several rules in mind:
- wires in Sketch must be closed
- wires in Sketch cannot touch with each other, because by FreeCAD such wires are considered as single wire
- this tool works also with many wires at one in Sketch, not only with single

If you look for stable FreeCAD version:
- you can download installation package with best stable FreeCAD 0.21.2 version: [magicCAD_0.21.AppImage](https://github.com/dprojects/Woodworking-package/releases/download/0.21/magicCAD_0.21.AppImage)
- and also I just released version with FreeCAD 1.0.2 kernel: [magicCAD_1.0.AppImage](https://github.com/dprojects/Woodworking-package/releases/download/1.0/magicCAD_1.0.AppImage) - however there are [several unsolved issue from FreeCAD 1.0](https://github.com/dprojects/Woodworking/issues/49).

I am going to close this issue, because I have already tested it again at [magicCAD_1.0.AppImage](https://github.com/dprojects/Woodworking-package/releases/download/1.0/magicCAD_1.0.AppImage) to be sure it is really working, so this issue might be rather related to unstable FreeCAD 1.1dev version and it is is rather not good idea to report issue based on such unstable environment...

accspert, 2025-08-13:
Hi Darwin, I use the stable version of freecad 1.0.1.

As I wrote in the comments it's not about wire2pad it always happens if you
open a sketch from the woodworking workbench.

dprojects, 2025-08-13:
> Hi Darwin, I use the stable version of freecad 1.0.1.

LOL, the FreeCAD 1.0.1 is obsolete from Darwin evolution perspective :-)

> As I wrote in the comments it's not about wire2pad it always happens if you
> open a sketch from the woodworking workbench.

it is rather not related to Woodworking workbench because there is nothing additional done during Sketch opening... moreover I have never came across such issue... so maybe this is operating system issue... 

Good Luck, you made my day :-)