dprojects/Woodworking #78: Creanting a lattice fence

atovidawood, 2025-09-13:
Hello,

I am trying to create a lattice fence, and I have some difficulties, the first problem is to rotate the crossbar and copy them in Z axis, I would like to make a 45º cut on both sides. Is a bit difficult to explain, because of that I will posted a Photo, the most problem is to copy in Z axis the crossbar, I want to make an chamfer on both side to adapt well the crossbar. If you could tell me how could I do that on the best way, I would apreciate it. Im learning wood working. Thanks.

[image]

dprojects, 2025-09-13:
Hi @atovidawood 

If I were you I would probably create the whole pattern as PartDesign object, and later use magicMove and Clones to create full fence. Probably I would create full pattern in Sketch to create correctly fit cut and have parametric piece. 

However, if you still want to create it from scratch like this, you may want to convert all the simple Part::Box objects into PartDesign::Pad objects via panel2pad tool and later edit the Sketches or just use chamfer PartDesign tool... also you can try [Router bit - chamfer](https://github.com/dprojects/Woodworking/tree/master/Docs#router-bit---chamfer) tools but this may not fit if the panel is not fit perfectly... 

In the past I wanted to show how to design garden patio with similar pattern but I had something more important to do, so this not happened. Now I have to create video tutorial about tenons, and after it I will look into "improving patterns"... so I will leave this issue open...

atovidawood, 2025-09-16:
> Hi again @atovidawood <https://github.com/atovidawood>
>
> I finally created the tutorial about tenons How to create tenon and
> mortise quickly <https://www.youtube.com/watch?v=vuddlHfAbCc> so now I
> have a little time to look into the fence pattern problem.
>
> I think you can create the pattern inside the container and clone the
> whole container. Thanks to this approach you can change the container
> content (pattern) later and the fence will be updated.
>
> I created sample here: Fence_pattern_v1.FCStd.zip
> (remove .zip)
> pattern_v1.png (view on web)
>
> In this example the vertical beams are only to create references for the
> middle beams. There is only single beam designed, the other one is mirror
> of the first one. The cut might be any angle, it depends on the vertical
> size. The beams can be connected with some kind of dowels, tenons, or even
> pocket holes. Also they might have half-lap joint in the middle (it is no
> designed here to avoid cyclic reference in SubShapeBinder to the Clone).
> Also if you resize, move for example to the right the right vertical beam
> the pattern will be resized, but in this case you have to glue the other
> columns of the pattern to follow resizing.
>
> Let me know if this is what you want.

I'm going to take a look at the file and I answer, I'm going to develop the
idea of the clearest lattice to see how it could be done in freecad, I
started with Freecad because I want to do a lot of projects at home and
most of them are with wood, that's why your wooden module is something
incredible, thank you very much for your work

atovidawood, 2025-09-25:
> Hi again @atovidawood <https://github.com/atovidawood>
>
> I just added new option
> <https://github.com/dprojects/Woodworking/tree/master/Docs#magicmove> to
> magic move for rotated objects. Now during copy you can change Object type
> into rotated and the object size will be calculated directly from
> getSizes function not from vertices. So the offset will be correct from the
> object placement.
>
> I think now will be more clear for everyone how to create copy with equal
> offset for rotated objects.
>
> I guess I can close this issue. Let me know, reopen this issue, if you
> still have problem with this kind of pattern.
>
> Thanks
> Darek

Thanks very much for the information